amanusk/s-tui #6: Clarify install process and move it at the top of documentation

svanoort, 2017-04-14:
Just to make it easy for people to figure out what to do. 

Even if it's just "install stress, then download and run the binary" (link here).

DonOregano, 2017-06-20:
You might add a comment about there being an AUR package for arch as of a few minutes ago. "s-tui-git" it is called.

amanusk, 2017-08-09:
Hope it is as simple possible now.

amanusk, 2017-08-17:
@DonOregano. A questions of someone not familiar enough with AUR, does the package need to be updated, or is it updated from Github?
Thanks.

DonOregano, 2017-08-22:
I've now replaced the AUR package with a version (named "s-tui" instead of "s-tui-git") that uses the released version instead of the git master. This package I need to bump when you release a new version.

amanusk, 2017-08-23:
@DonOregano. Thank you. I'll update the README about the new packaging name.  
I have recently uploaded version 0.5, and noticed that the version number of the AUR package did not change [https://aur.archlinux.org/packages/s-tui/](https://aur.archlinux.org/packages/s-tui/).  
I am not familiar enough with AUR packaging, so I wonder if it was supposed to update or is it a manual process.

closed 2019-04-01 by amanusk