amanusk/s-tui #9: Not able to run on Arch Linux

HarlemSquirrel, 2017-06-04:
I cloned the repo, installed urwid and psutil, and attempted to run `s-tui.py` as root. 

```
➤  sudo ./s-tui.py                                                                   1 ↵
Traceback (most recent call last):
  File "/home/hs/code/s-tui/aux.py", line 43, in read_msr
    read_res = f.read(8)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./s-tui.py", line 866, in <module>
    main()
  File "./s-tui.py", line 849, in main
    graph_controller = GraphController()
  File "./s-tui.py", line 779, in __init__
    self.view = GraphView(self)
  File "./s-tui.py", line 312, in __init__
    self.graph_data = GraphData(0)
  File "./s-tui.py", line 166, in __init__
    available_freq = read_msr(TURBO_MSR, 0)
  File "/home/hs/code/s-tui/aux.py", line 47, in read_msr
    e.message = e.message + "Unable to read file " + msr_file
AttributeError: 'OSError' object has no attribute 'message'
```

amanusk, 2017-06-04:
Thanks for the input.
Just to understand the problem, does it work well without sudo?
Does it work when trying to run it with sudo a second time?
Try running `/sbin/modprobe msr` before running with sudo.

HarlemSquirrel, 2017-06-04:
I do get the graph for a brief moment before it stops when running without sudo. 

```
➤  ./s-tui.py 
You are running without root permissions. Run as root for best results
Traceback (most recent call last):
  File "./s-tui.py", line 866, in <module>
    main()
  File "./s-tui.py", line 850, in main
    graph_controller.main()
  File "./s-tui.py", line 803, in main
    self.loop.run()
  File "/usr/lib/python3.6/site-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python3.6/site-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python3.6/site-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python3.6/site-packages/urwid/main_loop.py", line 715, in _loop
    alarm_callback()
  File "/usr/lib/python3.6/site-packages/urwid/main_loop.py", line 164, in cb
    callback(self, user_data)
  File "./s-tui.py", line 807, in animate_graph
    self.view.update_graph()
  File "./s-tui.py", line 378, in update_graph
    self.graph_util.set_y_label(self.get_label_scale(0, self.MAX_UTIL, y_label_size))
  File "./s-tui.py", line 441, in get_label_scale
    label = [int(min + i * (max - min) / label_cnt) for i in range(label_cnt + 1)]
TypeError: 'float' object cannot be interpreted as an integer
```

DonOregano, 2017-06-05:
I am able to run it with "python2 s-tui.py" (using the git version, have not tried the released version)

HarlemSquirrel, 2017-06-06:
I finally got it working with the following. Thanks for the help!
```
$ sudo pacman -S python2-pip
$ sudo pip2 install urwid
$ sudo pip2 install psutil
$ python2 s-tui.py
```

closed 2017-06-06 by HarlemSquirrel

amanusk, 2017-06-06:
Glad it worked out.
You can probably change the first line of s-tui.py `#!/usr/bin/python` to wherever python interpreter you are using. This way you can avoid adding `python2` every time. Or, make a soft link that links `/usr/bin/python` to what you have as python2.

By the way, were you having the same problem running the executable from releases?

HarlemSquirrel, 2017-06-06:
I didn't try the binary file.