amanusk/s-tui #12: Error running s-tui for first time

fsaft, 2017-06-22:
Hi, I get the following output in terminal after installing with pip:
`Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 9, in <module>
    load_entry_point('s-tui==0.2.3', 'console_scripts', 's-tui')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 542, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2569, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 26, in <module>
    from ComplexBarGraphs import ScalableBarGraph
ImportError: No module named 'ComplexBarGraphs'`
I am running Ubuntu 16.04.2 64bit

amanusk, 2017-06-22:
Could be related to python version. Could you try installing with pip2.7?
`pip2.7 install s-tui`
with sudo or --user at the end

fsaft, 2017-06-22:
This worked, thanks! Thread can be closed.

closed 2017-06-22 by fsaft

amanusk, 2017-06-27:
setup.py was updated to only work with python2.7
Installation with pip2.7 is suggested if default python is python3