amanusk/s-tui #16: Can't start s-tui

Niemand112233, 2017-07-15:
Hi,
everytime I try to start it (on two different server), I get this error:

sudo s-tui
Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/s_tui/s_tui.py", line 715, in main
    graph_controller.main()
  File "/usr/local/lib/python2.7/dist-packages/s_tui/s_tui.py", line 589, in main
    self.loop.run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 710, in _loop
    self._entering_idle()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 671, in _entering_idle
    callback()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 564, in entering_idle
    self.draw_screen()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 578, in draw_screen
    canvas = self._topmost_widget.render(self.screen_size, focus=True)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 225, in render
    canv = self._original_widget.render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 225, in render
    canv = self._original_widget.render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 566, in render
    canv = self._original_widget.render((maxcol,)+size[1:], focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/s_tui/ComplexBarGraphs.py", line 32, in render
    canvas = super(ScalableBarGraph, self).render(size, focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 547, in render
    disp = self.calculate_display((maxcol, maxrow))
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 390, in calculate_display
    maxrow)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 633, in calculate_bargraph_display
    segments = scale_bar_values(bar, top, maxrow)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 794, in scale_bar_values
    return [maxrow - int(float(v) * maxrow / top + 0.5) for v in bar]
ValueError: could not convert string to float:

amanusk, 2017-07-15:
The issue should have been resolved in the latest version. It was probably caused by temperature reading not being available.   
Try updating to the latest version by pulling from the git repo, or if you have installed via pip, running:
`sudo pip install --upgrade s-tui`

Niemand112233, 2017-07-15:
that worked! strange, I freshly installed it for the first time 7 hours ago...
sadly the temperature reading of AMD-CPUs doesn't work.

amanusk, 2017-07-15:
Great. Yes, sadly I had no chance to to try it on AMD hardware. But adding support for Additional temperature readings should not be too difficult. 
I am using psutil to read the CPU temperature. Could you run a python shell and enter the commands:
```
import psutil
psutil.sensors_temperatures()
````
And print out the output?

Another option is to make sure you have `lm-sensors` installed and run `sudo sensors` once.

Niemand112233, 2017-07-15:
lm-sensors is installed and the way I monitor it acutally. 

I tried it with the Python shell:
`>>> import psutil
>>> psutil.sensors_temperatures()
{'it8721': [shwtemp(label='', current=31.0, high=-123.0, critical=-123.0), shwtemp(label='', current=29.0, high=-84.0, critical=-84.0), shwtemp(label='', current=-128.0, high=-46.0, critical=-46.0)], 'k10temp': [shwtemp(label='', current=22.875, high=70.0, critical=90.0)]}
`
but it doesn't work. 

Now I get this error after exiting s-tui:
`
Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/s_tui/s_tui.py", line 718, in main
    graph_controller.main()
  File "/usr/local/lib/python2.7/dist-packages/s_tui/s_tui.py", line 590, in main
    self.loop.run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 703, in _loop
    ready, w, err = select.select(fds, [], fds, timeout)
`

amanusk, 2017-07-16:
Thanks for the help! I will add sensors it8721 to the next version. 
The error message you are getting is probably because you are quitting with Ctrl+C.  
I have added a signal handler to catch Ctrl+C and exit in a clean manner. Also all stress processes started by s-tui should be terminated if you exit with Ctrl+C starting from the next version.  
Again, thanks for the insights.

Niemand112233, 2017-07-16:
You're welcome. Let me know (best way will be writing here) if you made the update and I'll test it.

But a question is left: How should the programm be closed if not with Ctrl + C? I didn't see any exit button?

amanusk, 2017-07-16:
I have made the update. The 'Quit' button is on the side, right beneath the check box for selecting graphs. 'q' will also quit. Now I have added the option to close with Ctrl+C which is probably a popular choice.

Niemand112233, 2017-07-16:
ah, there did it hide :)

I upgraded and it works like a charm now :) 👍

amanusk, 2017-07-17:
Must be a different sensor. Can you post the results of psutil on this system?

closed 2017-07-21 by amanusk