amanusk/s-tui #22: Ryzen temperatures

abba23, 2017-08-16:
The temperature graph in s-tui is empty on my system, although I can see the temperatures using the it78 module and `sensors`.

OS: Arch Linux (kernel 4.12.6)
CPU: AMD Ryzen 7 1700X
Mainboard: ASUS Prime B350-Plus

amanusk, 2017-08-17:
Thanks for opening the issue. I didn't have a chance to test it on Ryzen platforms yet. The sensors for temperature vary slightly between motherboards and CPUs. I am updating them upon request. 
Could you please post the result of lm-sensors?

abba23, 2017-08-17:
Sure, thanks for doing this!

This is the normal output:
> it8655-isa-0290
> Adapter: ISA adapter
> in0:          +0.93 V  (min =  +2.32 V, max =  +2.08 V)  ALARM
> in1:          +2.48 V  (min =  +0.00 V, max =  +2.78 V)
> in2:          +2.01 V  (min =  +2.77 V, max =  +1.30 V)  ALARM
> in3:          +1.98 V  (min =  +1.21 V, max =  +1.25 V)  ALARM
> in4:          +1.98 V  (min =  +0.00 V, max =  +2.29 V)
> in5:          +1.90 V  (min =  +1.79 V, max =  +2.04 V)
> in6:          +1.98 V  (min =  +1.73 V, max =  +2.65 V)
> 3VSB:         +1.65 V  (min =  +2.50 V, max =  +2.59 V)  ALARM
> Vbat:         +1.62 V  
> +3.3V:        +1.66 V  
> fan1:        1541 RPM  (min =   11 RPM)
> fan2:           0 RPM  (min =   14 RPM)  ALARM
> fan3:        1201 RPM  (min =   10 RPM)
> temp1:        +47.0°C  (low  = +119.0°C, high = -35.0°C)  ALARM
> temp2:        +35.0°C  (low  =  -1.0°C, high = +125.0°C)
> temp3:        +37.0°C  (low  =  -1.0°C, high = -63.0°C)  ALARM  sensor = thermistor
> intrusion0:  ALARM
> 
> asus-isa-0000
> Adapter: ISA adapter
> cpu_fan:        0 RPM

This is the raw output:
> it8655-isa-0290
> Adapter: ISA adapter
> in0:
>   in0_input: 1.330
>   in0_min: 2.322
>   in0_max: 2.082
>   in0_alarm: 1.000
>   in0_beep: 0.000
> in1:
>   in1_input: 2.485
>   in1_min: 0.000
>   in1_max: 2.780
>   in1_alarm: 0.000
>   in1_beep: 0.000
> in2:
>   in2_input: 2.006
>   in2_min: 2.769
>   in2_max: 1.297
>   in2_alarm: 1.000
>   in2_beep: 0.000
> in3:
>   in3_input: 1.984
>   in3_min: 1.210
>   in3_max: 1.254
>   in3_alarm: 1.000
>   in3_beep: 0.000
> in4:
>   in4_input: 1.984
>   in4_min: 0.000
>   in4_max: 2.289
>   in4_alarm: 0.000
>   in4_beep: 0.000
> in5:
>   in5_input: 1.886
>   in5_min: 1.788
>   in5_max: 2.038
>   in5_alarm: 0.000
>   in5_beep: 0.000
> in6:
>   in6_input: 1.973
>   in6_min: 1.733
>   in6_max: 2.649
>   in6_alarm: 0.000
>   in6_beep: 0.000
> 3VSB:
>   in7_input: 1.657
>   in7_min: 2.496
>   in7_max: 2.594
>   in7_alarm: 1.000
>   in7_beep: 0.000
> Vbat:
>   in8_input: 1.624
> +3.3V:
>   in9_input: 1.657
> fan1:
>   fan1_input: 1591.000
>   fan1_min: 11.000
>   fan1_alarm: 0.000
>   fan1_beep: 0.000
> fan2:
>   fan2_input: 0.000
>   fan2_min: 14.000
>   fan2_alarm: 1.000
>   fan2_beep: 0.000
> fan3:
>   fan3_input: 1214.000
>   fan3_min: 10.000
>   fan3_alarm: 0.000
>   fan3_beep: 0.000
> temp1:
>   temp1_input: 48.000
>   temp1_max: -35.000
>   temp1_min: 119.000
>   temp1_alarm: 1.000
>   temp1_offset: 0.000
>   temp1_beep: 0.000
> temp2:
>   temp2_input: 35.000
>   temp2_max: 125.000
>   temp2_min: -1.000
>   temp2_alarm: 0.000
>   temp2_offset: 0.000
>   temp2_beep: 0.000
> temp3:
>   temp3_input: 38.000
>   temp3_max: -63.000
>   temp3_min: -1.000
>   temp3_alarm: 1.000
>   temp3_type: 4.000
>   temp3_offset: 0.000
>   temp3_beep: 0.000
> intrusion0:
>   intrusion0_alarm: 1.000
> 
> asus-isa-0000
> Adapter: ISA adapter
> cpu_fan:
>   fan1_input: 0.000

amanusk, 2017-08-17:
Wow. That is a lot of sensors.   
Which one of these is the more accurate one ?

> temp1: +47.0°C (low = +119.0°C, high = -35.0°C) ALARM
> temp2: +35.0°C (low = -1.0°C, high = +125.0°C)
> temp3: +37.0°C (low = -1.0°C, high = -63.0°C) ALARM sensor = thermistor

And also, if I can trouble you, can you run a python console and run the following commands:
```
import psutil
psutil.sensors_temperatures()
```
Just to make sure I am reading it from the correct place.

abba23, 2017-08-17:
temp1 is the CPU temperature
temp2 is the Mainboard temperature
temp3 is the CPU temperature - 10

This is the psutil output:
>{'it8655': [shwtemp(label='', current=52.0, high=-35.0, critical=-35.0), shwtemp(label='', current=38.0, high=125.0, critical=125.0), shwtemp(label='', current=42.0, high=-63.0, critical=-63.0)]}

amanusk, 2017-08-17:
Great, thanks. I pushed a fix now

abba23, 2017-08-17:
Thanks a lot!

I was just thinking that doing something like

```
import psutil

temperature = 0
temperatures = psutil.sensors_temperatures()
chips = list(temperatures.keys())

if len(chips) == 1:
	temperature = temperatures[chips[0]][0].current

print(temperature)
```

would work for many systems and you wouldn't have to add every chip manually like you're doing now.

amanusk, 2017-08-17:
You are right. This is the part of the code I like the least, because of the various options for the temperature read. The problem is that not all the sensors have the correct temperature readout in [chips[0]][0], but rather [chips[1]][0] or some other combination. (The system I am using for example gives a constant temperature of 27c if I use [chips[0]][0]. very optimistic). So currently I am ,inefficiently, asking  users what is the accurate temperature. 
But yes, this needs to be generalized in some way. Maybe a regex.

abba23, 2017-08-17:
For systems with only a single chip (`len(chips) == 1`) like mine this might be an appropriate fallback though.

Maybe something like this would work better:
```
import sensors
sensors.init()

def temp():

	temperature = 0

	for chip in sensors.iter_detected_chips():
		for value in chip:
			if value.label == 'temp1':
				return value.get_value()

print(temp())

sensors.cleanup()
```

amanusk, 2017-08-17:
I see. This is a similar concept to your first code. Again the problem is with the systems like mine, where sensors gives:
```
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +66.0°C  (high = +84.0°C, crit = +100.0°C)
Core 0:         +66.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:         +65.0°C  (high = +84.0°C, crit = +100.0°C)
Core 2:         +65.0°C  (high = +84.0°C, crit = +100.0°C)
Core 3:         +64.0°C  (high = +84.0°C, crit = +100.0°C)
```
The first reading `acpitz-virtual` is a wrong one. 
I will add your code as a fallback. It probably catches most of the common cases. 
Thanks for the input.

abba23, 2017-08-17:
You could add `if 'ISA' in chip.adapter_name` to circumvent that.

Anyway, thanks again for your help!

closed 2017-08-17 by abba23

amanusk, 2017-08-17:
Thank you!
I'll definitely improve that part of the code. The more people get to test this, the more bugs like this will show up.

amanusk, 2019-04-23:
s-tui should now show *all* available temperature sensors