amanusk/s-tui #24: Wrong temp shown

pete910, 2017-08-22:
On a ryzen 1700x with a gigabyte K7 board.

Works as intended but shows the incorrect CPU temps.

I am using the it87 driver from here https://github.com/groeck/it87 

Antergos(Arch based) fully updated

Here's a pic of what the temp should be compared to what the program reports

[image]

I can attach a raw sensor output if you need/helps.

amanusk, 2017-08-23:
Thank you for opening the issue. 
It would help to get the read form psutil.
Please try running the following commands in a python shell:
```
import psutil
psutil.sensors_temperatures()
```

pete910, 2017-08-23:
Output of above

`{'it8686': [shwtemp(label='', current=32.0, high=127.0, critical=127.0), shwtemp(label='', current=42.0, high=127.0, critical=127.0), shwtemp(label='', current=36.0, high=127.0, critical=127.0), shwtemp(label='', current=34.0, high=127.0, critical=127.0), shwtemp(label='', current=39.0, high=-117.0, critical=-117.0), shwtemp(label='', current=34.0, high=127.0, critical=127.0)], 'it8792': [shwtemp(label='', current=32.0, high=127.0, critical=127.0), shwtemp(label='', current=33.0, high=127.0, critical=127.0), shwtemp(label='', current=35.0, high=127.0, critical=127.0)]}
`

By the looks of it, it's reading the first value reported, value/temp 3 is the CPU on mine.

Not sure on how you can differentiate them as they don't seem labelled like within sensors output. 

sensor output.

```
sensors
it8686-isa-0a40                                                                                                                                                                                                                                               
Adapter: ISA adapter                                                                                                                                                                                                                                          
in0:          +1.39 V  (min =  +0.00 V, max =  +3.06 V)                                                                                                                                                                                                       
in1:          +1.94 V  (min =  +0.00 V, max =  +3.06 V)                                                                                                                                                                                                       
in2:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)                                                                                                                                                                                                       
in3:          +1.96 V  (min =  +0.00 V, max =  +3.06 V)                                                                                                                                                                                                       
in4:          +1.18 V  (min =  +0.00 V, max =  +3.06 V)                                                                                                                                                                                                       
in5:          +0.89 V  (min =  +0.00 V, max =  +3.06 V)                                                                                                                                                                                                       
in6:          +1.45 V  (min =  +0.00 V, max =  +3.06 V)                                                                                                                                                                                                       
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)                                                                                                                                                                                                       
Vbat:         +3.24 V                                                                                                                                                                                                                                         
CPU FAN:      740 RPM  (min =    0 RPM)                                                                                                                                                                                                                       
Case Fan:     390 RPM  (min =    0 RPM)                                                                                                                                                                                                                       
SysT1:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor                                                                                                                                                                                
Chipset/SB:   +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor                                                                                                                                                                                
CPU:          +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI                                                                                                                                                                                 
PICx16:       +34.0°C  (low  = +127.0°C, high = +127.0°C)                                                                                                                                                                                                     
VRM:          +39.0°C  (low  =  +0.0°C, high = -117.0°C)  sensor = thermistor                                                                                                                                                                                 
[... cut ...]
SYS/Pump 5:  1155 RPM  (min =   10 RPM)                                                                                                                                                                                                                       
PCIx8:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor                                                                                                                                                                                
Water:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
SysT2:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor

```

amanusk, 2017-08-23:
Thanks. 
For now, I could just make an 'if' to take the third sensor for 'it8686'. Question is whether it is always in that order.  
What is the effect of the it87 drivers? The sensors are only visible after the driver installation, or is it just to fix the offset?

pete910, 2017-08-23:
Not sure if it's the 3rd temp on all of the it8686 based to be honest. May be an option to chose temp reading used may be best bet. 
You need to force the default it87 that ships to use the it8686 chip thus using the groeck it87, also the shipped driver can hang some boards when probing the IO 

Offset isn't changed, has to be done with a compute config in your sensors.conf file

amanusk, 2017-08-23:
> May be an option to chose temp reading used may be best bet.
We are working on it :smile:  
I'll got with reading the third temperature reading for now.

pete910, 2017-08-23:
Awesome source 😄

closed 2017-08-24 by amanusk