amanusk/s-tui #26: Monitor fan speed

elazarr, 2017-08-22:
To complement the temperature monitoring, it would be nice to monitor the CPU fan speed (or other system fans). This can help correlating fan speed control issues with overheating problems.

amanusk, 2017-10-10:
Added, currently only on the sidebar.  A default sensors is selected, but selecting manually is also possible.

closed 2017-10-28 by amanusk

mdPlusPlus, 2019-08-06:
>selecting manually is also possible

I don't see an option for this in the interface. Can this be changed in the config file? By default it shows my GPU fan speed instead of the CPU one.
```
$ s-tui --version
s-tui 0.8.3 - (C) 2017-2018 Alex Manuskin, Gil Tsuker
    Released under GNU GPLv2
```

amanusk, 2019-08-07:
@mdPlusPlus, you should try the new version, it is currently available as a beta via pip:
[](https://pypi.org/project/s-tui/1.0.0b1) or by running from source.
It shows multiple fan sensors, and you can choose which sensors to show.

Saving settings is also persistent between session.

mdPlusPlus, 2019-08-08:
Your command got cut of. For everyone who might do the same: `pip3 install --pre --user s-tui`

amanusk, 2019-08-09:
Oops, thanks @mdPlusPlus,
Feedback on new interface appreciated :smile: