amanusk/s-tui #28: crashes after launch as root

l0ka, 2017-08-29:
After launching `s-tui` as root, it crashes after ~2 seconds, leaving the following output:
`Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/s_tui/s_tui.py", line 768, in main
    graph_controller.main()
  File "/usr/local/lib/python2.7/dist-packages/s_tui/s_tui.py", line 640, in main
    self.loop.run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 710, in _loop
    self._entering_idle()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 671, in _entering_idle
    callback()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 564, in entering_idle
    self.draw_screen()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 578, in draw_screen
    canvas = self._topmost_widget.render(self.screen_size, focus=True)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 225, in render
    canv = self._original_widget.render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 225, in render
    canv = self._original_widget.render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 566, in render
    canv = self._original_widget.render((maxcol,)+size[1:], focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/s_tui/ComplexBarGraphs.py", line 33, in render
    canvas = super(ScalableBarGraph, self).render(size, focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 547, in render
    disp = self.calculate_display((maxcol, maxrow))
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 390, in calculate_display
    maxrow)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 633, in calculate_bargraph_display
    segments = scale_bar_values(bar, top, maxrow)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 794, in scale_bar_values
    return [maxrow - int(float(v) * maxrow / top + 0.5) for v in bar]
ZeroDivisionError: float division by zero`
I am on Debian Stretch (9.1) 64bit, Python 2.7.13, Intel Xeon E3-1200 v3/4th Gen

amanusk, 2017-08-29:
Thank you for the bug report.
Does this only happen when running with root?

l0ka, 2017-08-29:
Yes, it happens only when I run it with root.

Indeed, I found another little problem, but I'm not already sure it's a
problem of your tool. The problem is that s-tui shows me the wrong cpu
model. I think it's because you use lscpu to detect the cpu model, but I'm
not sure, since I haven't read your code. The lspci command can detect the
right cpu model, but lscpu doesn't. I don't know why it's happening and I
didn't investigate, because I don't have enough time at the moment.

Thank you for your reply,
Alessio

amanusk, 2017-08-29:
I could think of 2 reasons
One is a permissions problem, could you please try running the executable directly?
`sudo /usr/local/bin/s-tui`
Or trying to clone and running the python script (
```
git clone https://github.com/amanusk/s-tui
cd s-tui
sudo ./s_tui/s_tui.py
```

The only other thing root is needed for is reading the max turbo frequency, this is read from an MSR. Try running  
`sudo modprobe msr`  

Regarding the other issue, I am reading the CPU name from /proc/cpuinfo

l0ka, 2017-08-30:
I tried both to run directly the executable as root and to clone and run
the python script as root. In both cases, it crashes with the same way as
before.
sudo modprobe msr gives no output.

amanusk, 2017-09-02:
The new version should fix this issue. Please update if the issue was resolved

l0ka, 2017-09-04:
Sorry for my late. I can confirm the issue is solved, thank you!

closed 2017-09-04 by l0ka