amanusk/s-tui #29: IOError: [Errno 61] No data available

sanderjo, 2017-08-30:
On my Ubuntu 17.04, I get a Traceback, with `IOError: [Errno 61] No data available`. This happens for the pip and the github version.
A `--help` works.

Any help / tips?

```
sander@Stream-13:~$ sudo s-tui 
Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 11, in <module>
    load_entry_point('s-tui==0.5.0', 'console_scripts', 's-tui')()
  File "/usr/local/lib/python2.7/dist-packages/s_tui/s_tui.py", line 767, in main
    graph_controller = GraphController(args)
  File "/usr/local/lib/python2.7/dist-packages/s_tui/s_tui.py", line 617, in __init__
    self.data = GraphData(is_admin=is_admin)
  File "/usr/local/lib/python2.7/dist-packages/s_tui/GraphData.py", line 60, in __init__
    self.rapl_power_reader = RaplPower()
  File "/usr/local/lib/python2.7/dist-packages/s_tui/RaplPower.py", line 24, in __init__
    self.max_power = self.read_max_power_file() / self.MICRO_JAUL_IN_JAUL
  File "/usr/local/lib/python2.7/dist-packages/s_tui/RaplPower.py", line 35, in read_max_power_file
    return float(self.read_measurement(self.intel_rapl_package_max_energy_file))
  File "/usr/local/lib/python2.7/dist-packages/s_tui/RaplPower.py", line 28, in read_measurement
    value = file.read()
IOError: [Errno 61] No data available
sander@Stream-13:~$ 
```
and

```
sander@Stream-13:~/git/s-tui$ sudo s_tui/s_tui.py 
Traceback (most recent call last):
  File "s_tui/s_tui.py", line 791, in <module>
    main() 
  File "s_tui/s_tui.py", line 767, in main
    graph_controller = GraphController(args)
  File "s_tui/s_tui.py", line 617, in __init__
    self.data = GraphData(is_admin=is_admin)
  File "/home/sander/git/s-tui/s_tui/GraphData.py", line 60, in __init__
    self.rapl_power_reader = RaplPower()
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 24, in __init__
    self.max_power = self.read_max_power_file() / self.MICRO_JAUL_IN_JAUL
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 35, in read_max_power_file
    return float(self.read_measurement(self.intel_rapl_package_max_energy_file))
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 28, in read_measurement
    value = file.read()
IOError: [Errno 61] No data available

```
and
```
sander@Stream-13:~/git/s-tui$ sudo s_tui/s_tui.py --help
usage: s_tui.py [-h] [-d] [-c] [-t] [-j] [-v]

********s-tui manual********
Usage in graphical mode:
* Toggle between stressed and regular operation using the radio buttons.
* If you wish to alternate stress defaults, you can do it in 'stress options
* If your system supports it, you can use the utf8 button to get a smoother graph
* Reset buttons resets the graph and the max statistics

optional arguments:
  -h, --help      show this help message and exit
  -d, --debug     Output debug log to _s-tui.log
  -c, --csv       Save stats to csv file
  -t, --terminal  Display a single line of stats without tui
  -j, --json      Display a single line of stats in JSON format
  -v, --version   Display version

```
and

```
sander@Stream-13:~/git/s-tui$ sudo s_tui/s_tui.py --terminal
Traceback (most recent call last):
  File "s_tui/s_tui.py", line 791, in <module>
    main() 
  File "s_tui/s_tui.py", line 767, in main
    graph_controller = GraphController(args)
  File "s_tui/s_tui.py", line 617, in __init__
    self.data = GraphData(is_admin=is_admin)
  File "/home/sander/git/s-tui/s_tui/GraphData.py", line 60, in __init__
    self.rapl_power_reader = RaplPower()
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 24, in __init__
    self.max_power = self.read_max_power_file() / self.MICRO_JAUL_IN_JAUL
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 35, in read_max_power_file
    return float(self.read_measurement(self.intel_rapl_package_max_energy_file))
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 28, in read_measurement
    value = file.read()
IOError: [Errno 61] No data available
sander@Stream-13:~/git/s-tui$ 
```

sanderjo, 2017-08-30:
I've put some debug printing into 
```
sander@Stream-13:~/git/s-tui$ sudo s_tui/s_tui.py 
('SJ: file_path', '/sys/class/powercap/intel-rapl/intel-rapl:0/energy_uj')
('SJ: file_path', '/sys/class/powercap/intel-rapl/intel-rapl:0/constraint_0_max_power_uw')
Traceback (most recent call last):
  File "s_tui/s_tui.py", line 791, in <module>
    main() 
  File "s_tui/s_tui.py", line 767, in main
    graph_controller = GraphController(args)
  File "s_tui/s_tui.py", line 617, in __init__
    self.data = GraphData(is_admin=is_admin)
  File "/home/sander/git/s-tui/s_tui/GraphData.py", line 60, in __init__
    self.rapl_power_reader = RaplPower()
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 24, in __init__
    self.max_power = self.read_max_power_file() / self.MICRO_JAUL_IN_JAUL
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 36, in read_max_power_file
    return float(self.read_measurement(self.intel_rapl_package_max_energy_file))
  File "/home/sander/git/s-tui/s_tui/RaplPower.py", line 29, in read_measurement
    value = file.read()
IOError: [Errno 61] No data available
```

OK, so:
```
('SJ: file_path', '/sys/class/powercap/intel-rapl/intel-rapl:0/energy_uj')
('SJ: file_path', '/sys/class/powercap/intel-rapl/intel-rapl:0/constraint_0_max_power_uw')
```
Let's check them:
```
sander@Stream-13:~/git/s-tui$ ll /sys/class/powercap/intel-rapl/intel-rapl:0/energy_uj
-r--r--r-- 1 root root 4096 aug 30 20:59 /sys/class/powercap/intel-rapl/intel-rapl:0/energy_uj
sander@Stream-13:~/git/s-tui$ cat /sys/class/powercap/intel-rapl/intel-rapl:0/energy_uj
106163220448
```

and:

```
sander@Stream-13:~/git/s-tui$ ll /sys/class/powercap/intel-rapl/intel-rapl:0/constraint_0_max_power_uw
-r--r--r-- 1 root root 4096 aug 30 20:59 /sys/class/powercap/intel-rapl/intel-rapl:0/constraint_0_max_power_uw
sander@Stream-13:~/git/s-tui$ cat /sys/class/powercap/intel-rapl/intel-rapl:0/constraint_0_max_power_uw
cat: '/sys/class/powercap/intel-rapl/intel-rapl:0/constraint_0_max_power_uw': No data available

```
... `No data available` doesn't look good, does it?

sanderjo, 2017-08-30:
I don't know if this is related: https://askubuntu.com/questions/930372/setting-intel-gpu-limits-in-ubuntu-17-04-no-longer-works/930449

sanderjo, 2017-08-31:
I checked on my Ubuntu 14.04 VPS ( Xeon E5335 ):

s-tui does run

/sys/class/powercap/ is empty ... so apparently s-tui can work without that information

```
sander@haring:~$ ll /sys/class/powercap/
total 0
drwxr-xr-x  2 root root 0 Aug 31 06:59 ./
drwxr-xr-x 49 root root 0 Aug 31 06:59 ../
sander@haring:~$
```

amanusk, 2017-08-31:
Thank you for the detailed information.  It really helps.  
I'll add a try catch for this case so it will not crash in case information is not available.  
 Also I'll try to see how getting the information is possible on 17.04

sanderjo, 2017-08-31:
> Also I'll try to see how getting the information is possible on 17.04

What does /sys/class/powercap/intel-rapl/intel-rapl:0/constraint_0_max_power_uw provide?

RAPL = Running Average Power Limit
... but what is `constraint_0_max_power_uw` ?

amanusk, 2017-08-31:
Theoretically, this should be the highest power draw.  But in the next version I'll just use the maximum recorded power draw per session.

sanderjo, 2017-09-01:
With a try/except in read_max_power_file() I got s-tui working:

With -1
```
    def read_max_power_file(self):
        if not self.is_available:
            return -1
        try:
            return float(self.read_measurement(self.intel_rapl_package_max_energy_file))
        except:
            return -1
```
it is working, but without a powerline:
[image]

With the value 2222222 micro-Joule (not "JAUL" ...) = 2 .2 J/s = 2.2 Watt, I get a useable powerline:
```
    def read_max_power_file(self):
        if not self.is_available:
            return -1
        try:
            return float(self.read_measurement(self.intel_rapl_package_max_energy_file))
        except:
            return 2222222
```

[image]

amanusk, 2017-09-01:
Thanks.It will be fixed in the upcoming version. Among many other things. The max_power will be dynamic according the the highest recorded power reading.

amanusk, 2017-09-02:
Please update whether the new version has fixed this issue.

sanderjo, 2017-09-03:
Yes, works on Ubuntu 17.04. Nice! 

About "Power":
- now in [W], which I like much better than te [J/s]
- power on my laptop is 18000 W ... that seems wrong

[image]

amanusk, 2017-09-03:
Seems like a scale problem. 18w / 57w could make sense. Leave this open for now.

sanderjo, 2017-09-03:
This laptop is tablet-style: Celeron (TDP 7W), eMMC (no disk), screen, no fan. So: low power consumption. 
The tool powertop "The battery reports a discharge rate of 4.58 W" ... that seems about right.

18 or 57 Watt seems unlikely to me.

amanusk, 2017-09-08:
Could you show the result of `cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj`

sanderjo, 2017-09-08:
Certainly:
```
sander@Stream-13:~$ cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj
131675646336
sander@Stream-13:~$ 
sander@Stream-13:~$ 
sander@Stream-13:~$ cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj
10664333056
sander@Stream-13:~$ 
sander@Stream-13:~$ 
sander@Stream-13:~$ cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj
21649881856
sander@Stream-13:~$ 
sander@Stream-13:~$ cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj
30872604544
sander@Stream-13:~$ 
```

amanusk, 2017-09-15:
```
741851264
4615478400
8544396032
11221446784
15208889280
19242829344
```
These readings are very strange, energy_uj should hold the energy in micro-joules. They shouldn't be changing so dramatically. I don't have an idea why is this so sporadic.

sanderjo, 2017-09-15:
Ah, "uj" = micro-Joule?

I tried again: value goes up and up, and then a rollover after 134258672512 = 1F4270C380 and after 132652438176 = 1EE2B392A0. So save to say: value grows to 1FFFFFFFF (=137438953471 = 2^37-1) That is 37 bits 

I'll update if I have more ideas.

```
sander@Stream-13:~$ while true; do cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj; sleep 1; done
71742470720
76885335520
80078047008
84503022368
89489333312
93804347328
100096093248
104374493792
108375847296
113604115296
117854383776
123638434432
126362131328
129743774848
134258672512
582493184
3892650848
8079320960
10567535552
13249817056
16887961312
19145489856
21722489376
24851344512
27078493568
31714767008
37248148320
41905850528
45147435168
49677220608
53394730240
56726185280
61558896160
65259616960
68739833088
73521948576
77626130240
80836133952
87490628384
91879771904
97222755328
103772152384
109235596640
114689206112
122660332640
127452004352
132652438176
1006201152
6640314336
12222023744
19994104160
25916703424
31845432384
38156132384
```

sanderjo, 2017-09-15:
If I do the delta between 1-sec-samples, and divice by 1 million, we get something steady & reasonable:

```
sander@Stream-13:~$ while true; do BLA=`cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj`; expr `expr $BLA - $OLD` / 1000111 ; OLD=$BLA; sleep 1; done
-30327
4124
3409
3294
3051
3820
4869
4161
5084
4213
4131
4244
5013
5296
4550
4041

```
As my system is a Celeron laptop, with TDP of 7 Watt, ... divide by 1000 once again:

```
sander@Stream-13:~$ while true; do BLA=`cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj`; expr `expr $BLA - $OLD` / 1000111222 ; OLD=$BLA; sleep 1; done
-54
4
3
3
3
3
2
2
2
3
5
5
4
4
4
4
4
4
4

```
... that looks rights: around 4 Watt.

If correct, I don't have micro-Joules, but nano-Joules ?

amanusk, 2017-09-15:
So it would seem.  
If it differs between systems there should probably be a scaling factor somewhere.

sanderjo, 2017-09-15:
It seems **Linux kernel dependent** :-1: 

I rebooted my Celeron laptop into Linux 4.10 (instead of 4.13), I now I get other results: the values grow less fast:

```
$ uname -a ; while true; do cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj; sleep 1; done
Linux Stream-13 4.10.0-32-generic #36-Ubuntu SMP Tue Aug 8 12:10:06 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
251549504
251835424
252140480
252445280
252726752
252983040
253236000
253483776
253731392
254029440
254309056
254549792
254787968
255056480

```
... but that seems too low: if I get the delta, and divide by 1 million, it's always 0:
```
$ uname -a; OLD=0; while true; do BLA=`cat /sys/class/powercap/intel-rapl/intel-rapl\:0/energy_uj`; expr `expr $BLA - $OLD` / 1000111 ; OLD=$BLA; sleep 1; done
Linux Stream-13 4.10.0-32-generic #36-Ubuntu SMP Tue Aug 8 12:10:06 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
329
0
0
0
0
0
0
0
0
```

And s-tui is now at 1 or 0 Watt:

[image]

sanderjo, 2017-09-15:
Linux 4.11 behalves like 4.10:

```
$ ./sj_get_energy_uh.sh 
Linux Stream-13 4.11.0-041100-generic #201705041534 SMP Thu May 4 19:36:05 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
76413280
76
76720224
0
77024544
0
77342304
0
77627552
0
77893504
0
78212896
0
78514464
0
78925920
0
79678720
```

amanusk, 2017-09-16:
Its worth testing 4.13 on other devices to see if perhaps only low power Celerons are effected.  
I am also considering implementing a different method of getting the power by directly reading the MSRs, but this requires root.

sanderjo, 2017-09-16:
Ouch, it's something else: not kernel related, but suspend related ...

This is too strange. Let's not try to solve it. 

After fresh start into Linux 4.13, the output was correct:

```
$ ./sj_get_energy_uh.sh 
Linux Stream-13 4.13.0-041300-generic #201709031731 SMP Sun Sep 3 21:33:09 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
195996128
195
196356512
0
196674976
0
197004704
0
197298336
0
197734528
0
198408960
0
199031200
0
199661792
0

```

On a second try also:

```
$ ./sj_get_energy_uh.sh 
Linux Stream-13 4.13.0-041300-generic #201709031731 SMP Sun Sep 3 21:33:09 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
205099680
205
205389824
0
205726080
0

```

But after a suspend (of only 10 seconds), I get the wrong output

```
$ ./sj_get_energy_uh.sh 
Linux Stream-13 4.13.0-041300-generic #201709031731 SMP Sun Sep 3 21:33:09 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
64985279808
64978
67683254784
2697
70513451872
2829
79373755584
8859
82126833024
2752
84931378592
2804

```

closed 2017-09-16 by sanderjo

amanusk, 2017-09-16:
Thank you for the help with this issue. 
I have pushed a small update to get the power read in the debug log. If it's not too much trouble, could you run it for a few seconds and print the log.  w/ and w/o stress.  
I am interested to see why in the correct case only 1/2 watts are shown.

sanderjo, 2017-09-16:
This is on my laptop which has suspended after reboot ... so big numbers.
[image]

BRB with a fresh boot

amanusk, 2017-09-16:
Thanks, 
I meant the output of _s-tui.log when running with the debug flag `python -m s_tui.s_tui -d`  
(After git pull)

sanderjo, 2017-09-16:
On 7th try, it started normally

[image]

amanusk, 2017-09-16:
Thanks.  
Looks like the reason for the crash is values between 0 and 1 read for power consumption (This might be correct for a low power cpu, my 35W TDP cpu idles at ~3w).   
Max power values should be rounded up to prevent the crash. I'll open a new issue to track this.

sanderjo, 2017-09-16:
FWIW: output of `sudo powerstat -d 2`

[image]

amanusk, 2017-09-16:
Yes. It shows total power based on battery drain. Makes sense that the CPU wiil ne ~1w