amanusk/s-tui #30: get_avarage_cpu_freq() gives "TypeError: unsupported operand type(s) for /: 'str' and 'int'" on 1-core VPS

sanderjo, 2017-09-01:
On my 1-core VPS (Ubuntu), s_tui does not show any frequency info, so I tried some things, which resulted in this Traceback
```
sander@haring:~/git/s-tui$ python s_tui/HelperFunctions.py
Traceback (most recent call last):
  File "s_tui/HelperFunctions.py", line 173, in <module>
    avg = get_avarage_cpu_freq()
  File "s_tui/HelperFunctions.py", line 40, in get_avarage_cpu_freq
    return round(reduce(lambda x, y: float(x) + float(y), cores_freq) / len(cores_freq), 1)
TypeError: unsupported operand type(s) for /: 'str' and 'int'
sander@haring:~/git/s-tui$ 
```

/proc/cpuinfo does contain frequency / MHz:
```
 cat /proc/cpuinfo | grep -i mhz
cpu MHz		: 1999.999
```

[image]

sanderjo, 2017-09-01:
Yeah, the single-core must be the problem: the reduce() with x and y won't work when the list cores_freq has only one entry. 

Some hacking: 

```
def get_avarage_cpu_freq():
    with open("/proc/cpuinfo") as cpuinfo:
        cores_freq = []
        for line in cpuinfo:
            if "cpu MHz" in line:
                core_freq = re.findall("\d+\.\d+", line)
                cores_freq += core_freq
        #print "SJ:", cores_freq, len(cores_freq)
        if len(cores_freq) == 1:
                return round(float(cores_freq[0]), 1)
        else:
                return round(reduce(lambda x, y: float(x) + float(y), cores_freq) / len(cores_freq), 1)

```

Now i have freq line, although only at '100' (not 2000 Mhz):
[image]

sanderjo, 2017-09-01:
... or less high-tech and thus easier to read:

```
def get_avarage_cpu_freq():
    with open("/proc/cpuinfo") as cpuinfo:
        cores = 0
        totalfreq = 0.0
        for line in cpuinfo:
            if "cpu MHz" in line:
                core_freq = re.findall("\d+\.\d+", line)[0]    # only first occurence, to avoid list
                cores += 1
                totalfreq += float(core_freq)
        return round(totalfreq/cores, 1)
```
Do you want a PR for this?

amanusk, 2017-09-01:
You are right, there is a problem with the single core.  
Try this instead:
``` python
        def get_avarage_cpu_freq():
            with open("/proc/cpuinfo") as cpuinfo:
                cores_freq = []
                for line in cpuinfo:
                    if "cpu MHz" in line:
                        core_freq = re.findall("\d+\.\d+", line)
                        cores_freq += core_freq
            return round(sum(float(x) for x in cores_freq) / len(cores_freq), 1)

```

sanderjo, 2017-09-01:
Yes, that works.

sanderjo, 2017-09-01:
FWIW but I tried this on system #3, and there is traceback (but the GUI is OK):
```
sander@nanopineo2:~/git/s-tui/s_tui$ python HelperFunctions.py
Traceback (most recent call last):
  File "HelperFunctions.py", line 172, in <module>
    avg = get_avarage_cpu_freq()
  File "HelperFunctions.py", line 40, in get_avarage_cpu_freq
    return round(sum(float(x) for x in cores_freq) / len(cores_freq), 1)
ZeroDivisionError: integer division or modulo by zero
```
Cause: the file `/proc/cpuinfo` of this ARM64 board has zero lines with "Hz" (ignore case) in it ... :-1: 
```
$ cat /proc/cpuinfo | grep -i hz
$
```

amanusk, 2017-09-02:
It would be difficult to test a different method without the hardware. If you have a different method of getting the frequency in this system, please try it and issue a PR.

sanderjo, 2017-09-03:
The current git version now works on my 1-core VPS:

FWIW: I now get
```
sander@haring:~/git/s-tui$ python s_tui/HelperFunctions.py
Traceback (most recent call last):
  File "s_tui/HelperFunctions.py", line 150, in <module>
    avg = get_avarage_cpu_freq()
NameError: name 'get_avarage_cpu_freq' is not defined
sander@haring:~/git/s-tui$ 
```

[image]

amanusk, 2017-09-03:
Good to hear. Thank you for your help in this.  
The error is only when running the module as main, the function was moved. Shouldn't be a problem.   
I'll close the issue.

closed 2017-09-03 by amanusk

sanderjo, 2017-09-03:
> If you have a different method of getting the frequency in this system, please try it and issue a PR.

I tried to find a way to find the freq, but so far no luck

FWIW: s_tui now works nice on my 4-core ARM64:
[image]