amanusk/s-tui #31: "ZeroDivisionError: float division by zero" when running with sudo, on 1-core VPS Ubuntu 14.04

sanderjo, 2017-09-01:
s_tui works, but says""
```
sander@haring:~/git/s-tui$ s_tui/s_tui.py
You are running without root permissions. Run as root to see max Turbo frequency
sander@haring:~/git/s-tui$
```
so I tried that, with a Traceback as result (after a 1 sec appearance of the ASCII Art) (see below).

The traceback is deep down in Urwid (Console user interface library for Python) ... so maybe it's a bug in there? 

```
sander@haring:~/git/s-tui$ sudo s_tui/s_tui.py
[sudo] password for sander:
Traceback (most recent call last):
  File "s_tui/s_tui.py", line 791, in <module>
    main()
  File "s_tui/s_tui.py", line 768, in main
    graph_controller.main()
  File "s_tui/s_tui.py", line 640, in main
    self.loop.run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 710, in _loop
    self._entering_idle()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 671, in _entering_idle
    callback()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 564, in entering_idle
    self.draw_screen()
  File "/usr/local/lib/python2.7/dist-packages/urwid/main_loop.py", line 578, in draw_screen
    canvas = self._topmost_widget.render(self.screen_size, focus=True)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 225, in render
    canv = self._original_widget.render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 225, in render
    canv = self._original_widget.render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/decoration.py", line 566, in render
    canv = self._original_widget.render((maxcol,)+size[1:], focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 1526, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 1751, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/container.py", line 2085, in render
    focus = focus and self.focus_position == i)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/sander/git/s-tui/s_tui/ComplexBarGraphs.py", line 33, in render
    canvas = super(ScalableBarGraph, self).render(size, focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/widget.py", line 141, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 547, in render
    disp = self.calculate_display((maxcol, maxrow))
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 390, in calculate_display
    maxrow)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 633, in calculate_bargraph_display
    segments = scale_bar_values(bar, top, maxrow)
  File "/usr/local/lib/python2.7/dist-packages/urwid/graphics.py", line 794, in scale_bar_values
    return [maxrow - int(float(v) * maxrow / top + 0.5) for v in bar]
ZeroDivisionError: float division by zero
sander@haring:~/git/s-tui$

```

amanusk, 2017-09-01:
This is the same as #28. The problem is probably not with Urwid. My guess is that the MSR used to get max frequency returns a 0. Using --debug should print the output of the MSR to examine the content.  
At any rate, a test to see if the returned frequency is > 0 will be added.

sanderjo, 2017-09-01:
OK, clear.

closed 2017-09-01 by sanderjo