amanusk/s-tui #32: Support for Power read on AMD systems

amanusk, 2017-09-10:
Power measurement is available on modern AMD systems via the fam15h_power sensor in 'lm_sensors'.  
Documentation is available here:   [fam15h_power](http://elixir.free-electrons.com/linux/latest/source/Documentation/hwmon/fam15h_power)  
An AMD system is needed to implement and test the feature.

jangrewe, 2018-11-16:
Hi, i came across this: https://github.com/djselbeck/rapl-read-ryzen
I had to load the `msr` kernel module, then it gave me pretty plausible wattage numbers for my TR 2950X. Would be great if you could somehow integrate it. :-)

amanusk, 2018-11-20:
@jangrewe thanks for the reference! This looks promising. 
I'll look into getting more information by reading MSRs. This will require root but would be a good option to have.

I hope I'll get a chance to implement this on Intel first (as I don't have AMD system), and it would be great if you could help test this on AMD (eventually)

schaerfo, 2020-07-22:
I would look into this based on the amd_energy driver (https://www.kernel.org/doc/html/v5.8-rc1/hwmon/amd_energy.html). This works at least for Ryzen 3000 series CPUs (I have one of those) but documentation suggests that Ryzen 1000 and 2000 series are also supported.

amanusk, 2020-07-22:
@schaerfo I don't have a system to test AMD(yet), if you have a working implementation, could you make a pull request?

zocker-160, 2020-08-15:
I am coming from the reddit post asking for help.

So here are my results using Kernel 5.8.1 on Ubuntu 20.04 and s-tui master:

[image]

amanusk, 2020-08-16:
@schaerfo I have asked some users on `/r/thinkpad` to check if AMD power measurements works on their system. Looks like something is missing as @zocker-160 shows.

Are there additional steps requires to enable this feature on the OS side, and make the relevant paths available? 
Thanks

schaerfo, 2020-08-16:
Thanks for the interest in this patch!
Unfortunately, since kernel 5.8.0-rc6, the `amd_energy` driver performs a more restrictive match for supported CPUs.
Therefore, it is necessary to either download the kernel source tree and revert commit `f28e360f29031fdef8df3a1bcad666243bd6872d` or use kernel 5.8.0-rc5 until this resolved in the kernel.

zocker-160, 2020-08-16:
>  it is necessary to either download the kernel source tree and revert commit f28e360f29031fdef8df3a1bcad666243bd6872d

@schaerfo @amanusk I can confirm that, I have tested the same with Kernel 5.8 and reverted the commit, which does indeed fix the problem:

[image]

amanusk, 2020-08-16:
@zocker-160 This is fantastic! Thanks!
@schaerfo do you have any idea if/when will this make it to the official kernel release?

schaerfo, 2020-08-16:
@amanusk Unfortunately not, the author of the `amd_energy` driver deliberately introduced this restriction in the mentioned patch citing "inconsistant values" being reported. For kernel 5.9 I have not found any changed in the driver. I will perhaps reach out to the developer to see if the restriction can be lifted somewhat. For that it would be useful to gather some data on which CPUs the reported numbers are reasonable.