amanusk/s-tui #35: "ImportError: No module named AboutMenu"

sanderjo, 2017-09-12:
After my "git pull" yesterday (and a fresh "git clone"0 I get "ImportError: No module named AboutMenu". 

```
sander@haring:~/git/s-tui$ git pull
Already up-to-date.
sander@haring:~/git/s-tui$ s_tui/s_tui.py
Traceback (most recent call last):
  File "s_tui/s_tui.py", line 40, in <module>
    from s_tui.AboutMenu import AboutMenu
  File "/home/sander/git/s-tui/s_tui/s_tui.py", line 40, in <module>
    from s_tui.AboutMenu import AboutMenu
ImportError: No module named AboutMenu
sander@haring:~/git/s-tui$
```

```
sander@haring:~/git/s-tui$ cd s_tui/
sander@haring:~/git/s-tui/s_tui$ ./s_tui.py
Traceback (most recent call last):
  File "./s_tui.py", line 40, in <module>
    from s_tui.AboutMenu import AboutMenu
  File "/home/sander/git/s-tui/s_tui/s_tui.py", line 40, in <module>
    from s_tui.AboutMenu import AboutMenu
ImportError: No module named AboutMenu
```

```
sander@haring:~/git/s-tui/s_tui$ ls -al AboutMenu.py
-rw-rw-r-- 1 sander sander 1972 Sep  1 07:43 AboutMenu.py
sander@haring:~/git/s-tui/s_tui$
```

amanusk, 2017-09-12:
I have added python3 support yesterday. This now happens when running s_tui.py directly. 
Running is still possible with `python -m s_tui.s_tui` from the main repo directory

sanderjo, 2017-09-12:
That works. Thanks!

closed 2017-09-12 by sanderjo