amanusk/s-tui #36: Running with -ct option results in 'Illegal sensor' and no temp readings

brokkr, 2017-09-14:
Version: 0.6.2
OS: Ubuntu 16.04
Installed using pip

Using the -ct option to bypass a virtual device reading, I get no temperature readings and a note in the debug log saying the sensor is 'Illegal'.

Here's my sensors output:

    acpitz-virtual-0
    Adapter: Virtual device
    temp1:        +27.8°C  (crit = +110.0°C)
    temp2:        +29.8°C  (crit = +110.0°C)
    
    coretemp-isa-0000
    Adapter: ISA adapter
    Package id 0:  +54.0°C  (high = +105.0°C, crit = +105.0°C)
    Core 0:        +54.0°C  (high = +105.0°C, crit = +105.0°C)
    Core 1:        +54.0°C  (high = +105.0°C, crit = +105.0°C)

s-tui defaults to the acpitaz-virtual-0 device. Seeing as this is always a steady 27.8 I figured it was not an actual reflection of the cpu temperature. Going on the --help instructions, I run s-tui in the following way.

     s-tui -ct coretemp,0

I have also tried `coretemp-isa` and `coretemp,1` and `coretemp,2`. In the program I get 'Temperature N/A'. In the log (running with debug on), I get the following:

    2017-09-14 09:41:03,713 [__init__()] [DEBUG]  stress is not installed
    2017-09-14 09:41:03,716 [__init__()] [DEBUG]  stress-ng is not installed
    2017-09-14 09:41:03,719 [__init__()] [INFO ]  num cpus 4
    2017-09-14 09:41:03,719 [__init__()] [DEBUG]  num cpus 4
    2017-09-14 09:41:03,720 [__init__()] [DEBUG]  [21, 21, 21, 21, 21, 21, 0, 0]
    2017-09-14 09:41:03,721 [update()] [INFO ]  Utilization recorded 26.3
    2017-09-14 09:41:03,722 [update()] [DEBUG]  MajorcoretempMinor0
    2017-09-14 09:41:03,722 [update()] [DEBUG]  Illegal sensor
    2017-09-14 09:41:03,722 [__init__()] [DEBUG]  Temperature sensor unavailable
    2017-09-14 09:41:03,732 [update()] [INFO ]  Utilization recorded 25.0
    2017-09-14 09:41:03,732 [update()] [DEBUG]  MajorcoretempMinor0
    2017-09-14 09:41:03,732 [update()] [DEBUG]  Illegal sensor
    ...

Explicitly setting the -ct option to acpitz result in the same thing just with acpitz in the log instead of coretemp.

amanusk, 2017-09-14:
Thank you for the detailed bug report. 
This is quite strange, coretemp should be the default.  
To test the source of this, could you please run a python shell and the following code and give the result:
``` python
import psutil
psutil.sensors_temperatures()
```

brokkr, 2017-09-14:
I ran it in both Python 2 and 3 as root. The result is the same:

    >>> psutil.sensors_temperatures()
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    AttributeError: 'module' object has no attribute 'sensors_temperatures'

    >>> psutil.__version__
    '3.4.2'

amanusk, 2017-09-14:
Thanks. The problem is your psutil version. Temperature sensors were only added  in a later version.  Upgrading it should solve the problem
```
sudo pip install psutil --upgrade
```
I'll see how I can force a newer psutil version during the installation

brokkr, 2017-09-14:
Upgrading to 5.3.1 solved it, thanks :-)

closed 2017-09-15 by amanusk