amanusk/s-tui #37: Crash on power read lower than 1W

amanusk, 2017-09-16:
On lower power systems, program crashes if power read is less than 1W

- [ ] Max Power read needs to be rounded up
- [ ] Side scale needs to show decimal point

amanusk, 2017-09-16:
@sanderjo 
I have pushed a fix to improve resolution of power read. This will probably not fix the issue of incorrect power read after suspend, but should display a more accurate graph when power is ~1 W

sanderjo, 2017-09-16:
After a `git pull` ... it looks nice:

[image]

amanusk, 2017-09-16:
Great! (BTW, a more accurate readout is available on the side bar if you scroll down).   
Does the problem persist after suspend?

sanderjo, 2017-09-16:
> Does the problem persist after suspend?

Yes: after a suspend and wake-up:

[image]

[image]

sanderjo, 2017-09-17:
What is the meaning of he file `/sys/class/powercap/intel-rapl/intel-rapl\:0/intel-rapl\:0\:0/energy_uj`, so with the extra `intel-rapl\:0\:0` in it ?
It seems to show more normal values (if I divide by 1000, not 1 million), also after a suspend. 

```
$ ./sj2.sh 
Linux Stream-13 4.13.0-041300-generic #201709031731 SMP Sun Sep 3 21:33:09 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
388804960
388804
388815648
10
388821376
5
388829728
8
388834944
5
388843808
8
388849248
5
388853856
4
388861312
7
388866336
5
388870624
4
```

amanusk, 2017-09-17:
intel-rapl\:0\:0 should be 'Cores' value and intel-rapl\:0\:1 is 'Uncore'  
intel-rapl\:0 should sum these together.

amanusk, 2017-09-27:
I'll close the issue, as there is no longer a crash.  
I'll open a new one to track irregular values after suspend.

closed 2017-09-27 by amanusk