amanusk/s-tui #39: Crash on opening STRESS option

MauroMombelli, 2017-09-24:
only different setting from original is enabling smooth graph. Arch linux, v0.6.2-1

```
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 11, in <module>
    load_entry_point('s-tui==0.6.2', 'console_scripts', 's-tui')()
  File "/usr/lib/python2.7/site-packages/s_tui/s_tui.py", line 628, in main
    graph_controller.main()
  File "/usr/lib/python2.7/site-packages/s_tui/s_tui.py", line 496, in main
    self.loop.run()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 710, in _loop
    self._entering_idle()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 671, in _entering_idle
    callback()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 564, in entering_idle
    self.draw_screen()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 579, in draw_screen
    self.screen.draw_screen(self.screen_size, canvas)
  File "/usr/lib/python2.7/site-packages/urwid/raw_display.py", line 748, in draw_screen
    for row in r.content():
  File "/usr/lib/python2.7/site-packages/urwid/canvas.py", line 595, in content
    sbody = shard_body(cviews, shard_tail)
  File "/usr/lib/python2.7/site-packages/urwid/canvas.py", line 931, in shard_body
    raise CanvasError("cviews do not fill gaps in"
urwid.canvas.CanvasError: cviews do not fill gaps in shard_tail!
```

amanusk, 2017-09-24:
Thank you for opening the issue.  
This is reproduced when the terminal running s-tui is too small. This needs to be fixed to work with any size window.

amanusk, 2017-09-26:
I have added a fix for the bug in the latest version. Tested all the menus in many window sized. Please update if this was fixed

closed 2017-10-10 by amanusk