amanusk/s-tui #40: Crash when running in Arch Linux TTY

amanusk, 2017-09-24:
The following traceback is presented:  
```
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 11, in <module>
    load_entry_point('s-tui==0.6.2', 'console_scripts', 's-tui')()
  File "/usr/lib/python2.7/site-packages/s_tui/s_tui.py", line 628, in main
    graph_controller.main()
  File "/usr/lib/python2.7/site-packages/s_tui/s_tui.py", line 496, in main
    self.loop.run()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python2.7/site-packages/urwid/main_loop.py", line 703, in _loop
    ready, w, err = select.select(fds, [], fds, timeout)
select.error: (9, 'Bad file descriptor')
```

mikiantena, 2017-09-24:
I get the same error on Arch Linux, which this was first reported for. Installed using yaourt from aur (s-tui-git package). On another machine I also tried the s-tui aur package (non-git), but the same error persists. I am also attaching installed packages list if it will help.

Also, this does **not** happen if I ssh into the same headless machine, from another X-windowed system. It doesn't actually matter if you are running X on the machine running s-tui. It only matters if you are running X on the machine you are currently working on.

[stuierror.log](https://github.com/amanusk/s-tui/files/1327525/stuierror.log)
[stuiinstalledpackages.txt](https://github.com/amanusk/s-tui/files/1327526/stuiinstalledpackages.txt)

amanusk, 2017-09-24:
Thank you for the input.  
I am unable to get this error when running s-tui from TTY without X server running in Ubuntu. So this could be Arch Linux specific. To test if the problem is with s-tui or with urwid, could you please try running some of urwid examples: https://github.com/urwid/urwid/tree/master/examples

amanusk, 2017-09-26:
Added option to open s-tui with --no-mouse option in latest release. The no-mouse flag will be passed to urwid.  
Please check if this fixed the problem.

mikiantena, 2017-09-26:
Hello. Thank you for you quick support and help.

I confirm that with the --no-mouse option (version 0.6.4) s-tui **works** perfectly in the above described situations. Using this option I am unable to replicate the problem again, and would regard this issue as resolved and fixed. Again, thank you.

amanusk, 2017-09-26:
That's great to hear. Thank you for your help with this.

closed 2017-09-26 by amanusk