amanusk/s-tui #41: Cpu temp on Ryzen (AB350m gaming3)

gsedej, 2017-09-25:
First: Thanks for this very useful tool!

The temperature reporting is not correct on my system.
First of all, the actual cpu "die" temperature is not yet mainline as of kernel 4.14, but there is path that can be applied. I installed the patched kernel and can see temp trough "k10temp-pci-00c3" in "sensors", but s-tui does still not report correct temperature

The motherboard has it87 (also need to patch from https://github.com/groeck/it87). Looks like the s-tui takes "temp1" from it87, but "temp1" look like its always ~28-31°C, probably measures air temperature in case.

amanusk, 2017-09-26:
Thank you for the positive feedback.  
It is possible to select a temperature sensors manually when starting s-tui. See the s-tui --help for and example.  
I have also added an option to select a sensor from the 'Temp Sensors' menu in the TUI. The new feature is in now in the master branch and pip package. Please update if this new feature works for you.

gsedej, 2017-09-26:
Thank you for reply. Looks like I had few weeks old pip version, so after updating I did get command line option and menu options for "Temp Sensors".

Now I can select "k10temp,0," to monitor actual cpu temperature. Notice the last comma - it should be there, without it, it fails to select sensor. Here i my sensors output for k10temp (using ubuntu 16.04)
```
$ sensors
k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +60.8°C  (high = +70.0°C)
```

Also I noticed that the temp graph has minimum at 0.0°C, which makes qute small differnece between

The ryzen cpus at the moment can't be yet monitored via "cpu die" temp, but when it gets mainlined, will the s-tui be able to select "k10temp,0," sensor by default? Or app simply cannot know which sensor is cpu and guesses one (temp1)?

amanusk, 2017-09-26:
> Notice the last comma - it should be there, without it, it fails to select sensor. 
Thanks for noticing that. I'll fix it.  

> Also I noticed that the temp graph has minimum at 0.0°C, which makes qute small differnece between 
Difference between?

> Or app simply cannot know which sensor is cpu and guesses one (temp1)?
On most Intel systems, the CPU temp sensor is known. On AMD I could not establish a clear cut way to know what the CPU sensors is. I have added some "correct" sensors for AMD systems when an issue was opened.  
So yes, a default is selected and is not always correct. Hence the option to select a sensor manually. If there is a deterministic way to know which sensors is the CPU die sensor on AMD systems, I'll be happy to hear about it.

gsedej, 2017-09-26:
> Difference between?
Sorry, forget to finish sentence. if the "idle" temp is 50C and max is 70C, the difference in graph height after stress is ran would be 20/70, because 0 is minimum. If minimum would be e.g. 40°C, the difference in height would be 20/30.  The graph would grow more. I hope I was able to explain.

So the temp sesnor for ryzen, threadripper and epyc will be k10temp, I will report when it will be mainlined.

gsedej, 2017-09-26:
instead of min, you can read min temp of cpu and round it to 10 or 5 degree (e.g. min=48C -> set graph to 45)
I will report output temperatures when I'm home.

gsedej, 2017-09-26:
here is the output:
```
$ ipython
Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
Type "copyright", "credits" or "license" for more information.

IPython 2.4.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import psutil

In [2]: psutil.sensors_temperatures()
Out[2]: 
{'it8792': [shwtemp(label='', current=26.0, high=127.0, critical=127.0),
  shwtemp(label='', current=30.0, high=127.0, critical=127.0),
  shwtemp(label='', current=30.0, high=127.0, critical=127.0)],
 'k10temp': [shwtemp(label='', current=51.875, high=70.0, critical=70.0)],
 'radeon': [shwtemp(label='', current=34.0, high=120.0, critical=120.0)]}

```

amanusk, 2017-09-30:
Leaving open until validated.

asim-vax, 2018-02-01:
installed 4.15 kernel (ubuntu) install lm sensors/sensors-detect, showed nothing - but both psensors and s-tui picked up the CPU temperature - with idle and load temperatures looking correct(matching the numbers i get from a windows boot) - One happy ryzen user here!

gsedej, 2018-02-16:
I can confirm it works with 4.15 (and also 4.15rc1). Closing bug.

closed 2018-02-16 by gsedej

amanusk, 2018-02-17:
@gsedej Thanks for your help with this.

commit 397e17fe (2017-09-30): Add try catch to temp sensors menu creation

+ Add k10temp to default temperature sensors. Should fix #41
+ Add IOError exception to sensors menu to fix #43