amanusk/s-tui #43: Crash reading sensors

MauroMombelli, 2017-09-30:
Hello, i was going to test the fix from #39 but not as soon as i start the program i get the error (s-tui 0.6.4)

```
s-tui
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 11, in <module>
    load_entry_point('s-tui==0.6.4', 'console_scripts', 's-tui')()
  File "/usr/lib/python2.7/site-packages/s_tui/s_tui.py", line 651, in main
    graph_controller = GraphController(args)
  File "/usr/lib/python2.7/site-packages/s_tui/s_tui.py", line 494, in __init__
    self.view = GraphView(self, args)
  File "/usr/lib/python2.7/site-packages/s_tui/s_tui.py", line 200, in __init__
    self.temp_sensors_menu = TempSensorsMenu(self.on_sensors_menu_close)
  File "/usr/lib/python2.7/site-packages/s_tui/TempSensorsMenu.py", line 65, in __init__
    sensors_dict = psutil.sensors_temperatures()
  File "/usr/lib/python2.7/site-packages/psutil/__init__.py", line 2268, in sensors_temperatures
    rawdict = _psplatform.sensors_temperatures()
  File "/usr/lib/python2.7/site-packages/psutil/_pslinux.py", line 1151, in sensors_temperatures
    current = float(cat(base + '_input')) / 1000.0
  File "/usr/lib/python2.7/site-packages/psutil/_pslinux.py", line 299, in cat
    return f.read().strip()
IOError: [Errno 22] Invalid argument
```

amanusk, 2017-09-30:
Thanks.  
It could be a problem with pstuil. Could you please run the following in a python shell.  
```python
import psutil
psutil.__version__
psutil.sensors_temperatures()
```

MauroMombelli, 2017-09-30:
```
python2
Python 2.7.14 (default, Sep 20 2017, 01:25:59) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> psutil.__version__
'5.3.1'
>>> psutil.sensors_temperatures()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/psutil/__init__.py", line 2268, in sensors_temperatures
    rawdict = _psplatform.sensors_temperatures()
  File "/usr/lib/python2.7/site-packages/psutil/_pslinux.py", line 1151, in sensors_temperatures
    current = float(cat(base + '_input')) / 1000.0
  File "/usr/lib/python2.7/site-packages/psutil/_pslinux.py", line 299, in cat
    return f.read().strip()
IOError: [Errno 22] Invalid argument
>>> 
```

so probably it is missing some dependency os-side?

MauroMombelli, 2017-09-30:
sensors seems to be installed and running:

```
sensors
radeon-pci-0100
Adapter: PCI adapter
temp1:            N/A  (crit = +120.0°C, hyst = +90.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +57.0°C  (high = +84.0°C, crit = +100.0°C)
Core 0:        +56.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:        +51.0°C  (high = +84.0°C, crit = +100.0°C)
Core 2:        +51.0°C  (high = +84.0°C, crit = +100.0°C)
Core 3:        +50.0°C  (high = +84.0°C, crit = +100.0°C)

acpitz-virtual-0
Adapter: Virtual device
temp1:        +57.0°C  (crit = +98.0°C)

```

amanusk, 2017-09-30:
If am to guess, probably there is a problem parsing "N/A" of radeon-pci.  
I'll add

amanusk, 2017-09-30:
@MauroMombelli could you please check if you are getting any temperature reading at all with the current master branch?

MauroMombelli, 2017-10-01:
it works, also taken ownership of the s-tui-git package on arch-linux AUR.

closed 2017-10-01 by amanusk

amanusk, 2017-10-02:
Thank you for your help with the debug.  
Does the s-tui-git package follow the master branch here, or does it need to be manually updated by you occasionally?

MauroMombelli, 2017-10-03:
None of them; when you force the package to reinstall, it will clone
current master

commit 397e17fe (2017-09-30): Add try catch to temp sensors menu creation

+ Add k10temp to default temperature sensors. Should fix #41
+ Add IOError exception to sensors menu to fix #43