amanusk/s-tui #47: pkg_resources.DistributionNotFound

GM-Script-Writer-62850, 2017-10-29:
Installed via PPA with xubuntu 17.04 64bit (i should upgrade soon, but i worked on my rpi the last 2 weekends)
had the issue where pkg_resource was not found, fixed it via `sudo apt-get install --reinstall python-pkg-resources`
it looks like psutil 5.0.1-1 is too old, but in the readme it says some features will not work, not that the application will not work at all, it looks like 17.10 has the same version as 17.04 

- https://packages.ubuntu.com/search?keywords=python-psutil

maybe you can provide 5.2 from your ppa?
```
chad@A54C-NB91:~$ python3 /usr/bin/s-tui
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3019, in <module>
    @_call_aside
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3032, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 655, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 849, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 's-tui==0.6.7' distribution was not found and is required by the application
```
```
chad@A54C-NB91:~$ python2 /usr/bin/s-tui
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3019, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3032, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 657, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 670, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 849, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'psutil>=5.2.0' distribution was not found and is required by s-tui
```

sojusnik, 2017-10-29:
Same issue under Ubuntu 17.10 here.

amanusk, 2017-11-02:
Thanks for opening the issue. I'll see if I can add pstuil in my PPA, if not, I'll remove the demand for psutil version >= 5.2.

amanusk, 2017-11-04:
It appears supporting and validating multiple packages for Ubuntu proves quite a handful.  
Best to simply update installation process to first install pip and install s-tui from pip.  
If someone wishes to maintain a PPA for s-tui they are more than welcome to do so. I would happily include it in the README

amanusk, 2018-06-07:
I have updated the PPA for 16.04 and 18.04 to version 0.7.6
Installation process for Ubuntu is now:
```
sudo add-apt-repository ppa:amanusk/python-s-tui
sudo apt-get update
sudo apt-get install python3-s-tui
```

deerainw, 2019-03-05:
Same issue, how to solve this on windows ?

closed 2020-07-11 by amanusk