amanusk/s-tui #48: Add support for saving settings (stress options, refresh time, etc)

Hermanio, 2017-11-02:
I find myself constantly changing the refresh time and graph smoothness settings. 

It would be nice to have support for a basic config file with those settings in the users home folder.
Alternatively, extra arguments for those settings and others, too (stress test settings and such), would be great and can easily be aliased.

amanusk, 2017-11-04:
That is definitely a good idea. Hopefully, will be implemented in next version. 

- [ ] Select smooth
- [ ] Select shown sensors
- [ ] Refresh rate
- [ ] Select sensors shown in sidebar

amanusk, 2018-04-29:
@Hermanio support for saving settings was added in the latest version. After pressing the `save settings` button, a config file will be saved and the same config should be restored the next time you start s-tui. 
Please tell me if this works as expected.

Hermanio, 2018-04-29:
Gave it a go, so far the refresh time, Smooth Graph option and temperature sensor preference are saved. Thank you for maintaining this project, has been a great help in debugging and building two tools of my own!

amanusk, 2018-04-29:
Great to hear. I'll close the issue, feel free to open a new one if this introduced a bug.

closed 2018-04-29 by amanusk