amanusk/s-tui #49: Crashes at launch on macOS: psutil doesn't have sensors_temperatures

williamleuschner, 2017-11-08:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel core i5, MacBook Air 6,2
  * OS version: macOS 10.13.1
  * s-tui version: 0.6.7
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
`s-tui` crashes immediately after launch with the below traceback.

#### Observed Results:
```
  Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/site-packages/s_tui/s_tui.py", line 679, in main
    graph_controller = GraphController(args)
  File "/usr/local/lib/python3.6/site-packages/s_tui/s_tui.py", line 522, in __init__
    self.view = GraphView(self, args)
  File "/usr/local/lib/python3.6/site-packages/s_tui/s_tui.py", line 207, in __init__
    self.temp_sensors_menu = TempSensorsMenu(self.on_sensors_menu_close)
  File "/usr/local/lib/python3.6/site-packages/s_tui/TempSensorsMenu.py", line 67, in __init__
    sensors_dict = psutil.sensors_temperatures()
AttributeError: module 'psutil' has no attribute 'sensors_temperatures'
```

#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 

```
2017-11-08 10:14:58,334 [main()] [INFO ]  Started without root permissions
2017-11-08 10:14:58,340 [__init__()] [DEBUG]  stress-ng is not installed
```
  
### Step 3: Reproduce the problem:
Occurs without fail at program launch on macOS.

#### Steps to reproduce:
1. `pip install s-tui`
2. `brew install stress`
3. `s-tui`

amanusk, 2017-11-08:
Thank you for the thorough bug report. 
The issue has been patched [here](https://github.com/amanusk/s-tui/blob/d55c66736e964a7c44eac3a2e879f4d29015ff1d/s_tui/TempSensorsMenu.py#L68) but it is not yet in the latest release version.  You are welcome to try cloning and running s-tui from source in the mean time. 
The bigger problem is that temperature reading for macOS is not yet available #27, and I am currently unable to address the issue without a working mac box (I am working on it).

amanusk, 2017-11-09:
A fix was uploaded to PyPi.  
Please try updating s-tui with `pip install s-tui --upgrade`

aanari, 2017-12-10:
It still fails on Mac OSX, here is the stack trace (I followed the same instructions as @williamleuschner):

```
$ s-tui
Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 11, in <module>
    load_entry_point('s-tui==0.6.7.2', 'console_scripts', 's-tui')()
  File "/usr/local/lib/python2.7/site-packages/s_tui/s_tui.py", line 679, in main
    graph_controller = GraphController(args)
  File "/usr/local/lib/python2.7/site-packages/s_tui/s_tui.py", line 522, in __init__
    self.view = GraphView(self, args)
  File "/usr/local/lib/python2.7/site-packages/s_tui/s_tui.py", line 216, in __init__
    urwid.WidgetPlaceholder.__init__(self, self.main_window())
  File "/usr/local/lib/python2.7/site-packages/s_tui/s_tui.py", line 464, in main_window
    temp_source = TemperatureSource(self.custom_temp)
  File "/usr/local/lib/python2.7/site-packages/s_tui/Sources/TemperatureSource.py", line 43, in __init__
    self.update() # Initial update
  File "/usr/local/lib/python2.7/site-packages/s_tui/Sources/TemperatureSource.py", line 75, in update
    last_value = psutil.sensors_temperatures()['coretemp'][0].current
AttributeError: 'module' object has no attribute 'sensors_temperatures'
```

amanusk, 2017-12-10:
@aanari Thanks for the update. 
An AttributeError exception was added to both temperature and fan sensors. This should prevent the crash. These metrics are not yet available in psutil for macOS but are hopefully coming soon.  [psutil-#371](https://github.com/giampaolo/psutil/issues/371#issuecomment-274963090).

shadowwalkersteam, 2021-02-15:
It is still happening. I'm also getting the same exception on macOS.

wnz99, 2022-05-12:
Getting the same error on macOS.