amanusk/s-tui #50: pressing escape to exit a popup makes s-tui crash

nixomose, 2017-12-10:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): _intel x86_64____
  * OS version: _ubuntu 16.04____
  * s-tui version: _0.6.7.2_
  * Installation method(pip/PPA/source code): ___pip__
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):
open up the help window, hit escape once or twice, then click help again  

Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 9, in <module>
    load_entry_point('s-tui==0.6.7.2', 'console_scripts', 's-tui')()
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 680, in main
    graph_controller.main()
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 544, in main
    self.loop.run()
  File "/usr/local/lib/python3.5/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/local/lib/python3.5/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/local/lib/python3.5/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/local/lib/python3.5/dist-packages/urwid/main_loop.py", line 719, in _loop
    self._watch_files[fd]()
  File "/usr/local/lib/python3.5/dist-packages/urwid/raw_display.py", line 393, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/local/lib/python3.5/dist-packages/urwid/raw_display.py", line 466, in parse_input
    codes, wait_for_more)
  File "/usr/local/lib/python3.5/dist-packages/urwid/escape.py", line 385, in process_keyqueue
    if run[0] == "esc" or run[0].find("meta ") >= 0:
AttributeError: 'tuple' object has no attribute 'find'

#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
* Replace this with debug output *
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. _____
2. _____
3. _____

```
* Replace this with relevant code to reproduce the problem *
```

amanusk, 2017-12-10:
Thank you for opening a bug report, 
Are you able to reproduce this behaviour? 
I have tried reproducing it with ubuntu 16.04 without success:
Pressing `esc` with the help menu opened and opening it again multiple times.
Are there any additional steps

nixomose, 2017-12-10:
maybe I wasn't clear about the mouse part:

1) click on the help button with the mouse
2) hit the esc key twice (the first time doesn't make it go away)
3) click the help button with the mouse again.
4) boom.

```
Traceback (most recent call last):_______________________________________________________________________________________________________________________________________________________│
  File "/usr/local/bin/s-tui", line 9, in <module>                                                Temperature [C]                                                                        │
    load_entry_point('s-tui==0.6.7.2', 'console_scripts', 's-tui')()                                                                                                                     │
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 680, in main                                                                                                        │
    graph_controller.main()                                                                                                                                                              │
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 544, in main                                                                                                        │
    self.loop.run()    │                                                                                                                                                                 │
  File "/usr/local/lib/python3.5/dist-packages/urwid/main_loop.py", line 278, in run                                                                                                     │
    self._run()        │                                                                                                                                                                 │
  File "/usr/local/lib/python3.5/dist-packages/urwid/main_loop.py", line 376, in _run                                                                                                    │
    self.event_loop.run()                                                                                                                                                                │
  File "/usr/local/lib/python3.5/dist-packages/urwid/main_loop.py", line 682, in run                                                                                                     │
    self._loop()       │ ________________________________________________________________________________________________________________________________________________________________│
  File "/usr/local/lib/python3.5/dist-packages/urwid/main_loop.py", line 719, in _loop               Power [W]                                                                           │
    self._watch_files[fd]().0                                                                                                                                                            │
  File "/usr/local/lib/python3.5/dist-packages/urwid/raw_display.py", line 393, in <lambda>                                                                                              │
    event_loop, callback, self.get_available_raw_input())                                                                                                                                │
  File "/usr/local/lib/python3.5/dist-packages/urwid/raw_display.py", line 466, in parse_input                                                                                           │
    codes, wait_for_more)                                                                                                                                                                │
  File "/usr/local/lib/python3.5/dist-packages/urwid/escape.py", line 385, in process_keyqueue                                                                                           │
    if run[0] == "esc" or run[0].find("meta ") >= 0:                                                                                                                                     │
AttributeError: 'tuple' object has no attribute 'find'         ```

amanusk, 2017-12-10:
Could be terminal related?
What terminal emulator are you using? Could you test this a differnt one and see if it still occurs?

nixomose, 2017-12-10:
well I was doing it in screen when it failed, so I tried it in a gnome terminal (I use xfce) and it was harder to get it to do it, but I just went all spazzy clicking on help and hitting escape, and eventually it failed.
If it is just me, don't worry about it.

amanusk, 2018-04-29:
@nixomose I have added a fix for this bug in the latest version. 
Please try upgrading with `pip install s-tui --upgrade` and see if the issue persists.
Thanks.

nixomose, 2018-05-03:
that seems to have fixed it, thanks.

closed 2018-05-03 by nixomose