amanusk/s-tui #52: AttributeError with Ubuntu

XavCC, 2018-01-29:
### Environment

  * System : Linux version 4.13.0-32-generic (buildd@lgw01-amd64-016) (gcc version 7.2.0 (Ubuntu 7.2.0-8ubuntu3)),  CPU: Intel Core i3-2370M @ 4x 2.4GHz
  * OS version: Ubuntu 17.10 artful
  * s-tui version: 0.6.7
  * Installation method : 
```
sudo add-apt-repository ppa:amanusk/python-s-tui
sudo apt-get update
sudo apt-get install python-s-tui
```
Option
```
sudo pip install urwid
sudo pip install psutil
sudo apt-get install stress
```
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):
```
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 9, in <module>
    load_entry_point('s-tui==0.6.7', 'console_scripts', 's-tui')()
  File "/usr/lib/python2.7/dist-packages/s_tui/s_tui.py", line 679, in main
    graph_controller = GraphController(args)
  File "/usr/lib/python2.7/dist-packages/s_tui/s_tui.py", line 522, in __init__
    self.view = GraphView(self, args)
  File "/usr/lib/python2.7/dist-packages/s_tui/s_tui.py", line 207, in __init__
    self.temp_sensors_menu = TempSensorsMenu(self.on_sensors_menu_close)
  File "/usr/lib/python2.7/dist-packages/s_tui/TempSensorsMenu.py", line 67, in __init__
    sensors_dict = psutil.sensors_temperatures()
AttributeError: 'module' object has no attribute 'sensors_temperatures'
```
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2018-01-29 16:15:11,022 [main()] [INFO ]  Started without root permissions
2018-01-29 16:15:11,030 [__init__()] [DEBUG]  stress-ng is not installed
```
  
### Step 3: Trying to solve the problem:
#### Steps :
1. `sudo apt install stress-ng`
2. `sudo s-tui`
3. Same AttributeError 

```
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 9, in <module>
    load_entry_point('s-tui==0.6.7', 'console_scripts', 's-tui')()
  File "/usr/lib/python2.7/dist-packages/s_tui/s_tui.py", line 679, in main
    graph_controller = GraphController(args)
  File "/usr/lib/python2.7/dist-packages/s_tui/s_tui.py", line 522, in __init__
    self.view = GraphView(self, args)
  File "/usr/lib/python2.7/dist-packages/s_tui/s_tui.py", line 207, in __init__
    self.temp_sensors_menu = TempSensorsMenu(self.on_sensors_menu_close)
  File "/usr/lib/python2.7/dist-packages/s_tui/TempSensorsMenu.py", line 67, in __init__
    sensors_dict = psutil.sensors_temperatures()
AttributeError: 'module' object has no attribute 'sensors_temperatures'

```

amanusk, 2018-01-30:
Hi @XavCC , Thank you for the detailed bug report. 
This issue should be fixed in the newest version . Unfortunately I currently lack the time and skill to keep updated PPAs for all Ubuntu releases.  The latest version with the fix is available via pip. 
```
sudo pip install s-tui
```
If you have installed via PPA, you should uninstal the old version first to avoid possible conflicts 
Please update if this fixes the problem for you

XavCC, 2018-03-21:
Thanks a lot @amanusk ;-)

it works with `sudo su`
```
sudo pip install s-tui
s-tui
```

Now I 'll try under Kali linux

closed 2018-04-04 by XavCC