amanusk/s-tui #53: Cant start s-tui with sudo (python3)

gsedej, 2018-02-16:
On Ubuntu 16.04, s-tui (from "pip") ran with sudo gives and error. 
When running with python2.7 it works (but the interface is different - menu on right)

error
```
sudo s-tui
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/s_tui/Sources/FreqSource.py", line 47, in read_msr
    read_res = f.read(8)
OSError: [Errno 5] input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/s_tui/Sources/FreqSource.py", line 76, in __init__
    available_freq = read_msr(TURBO_MSR, 0)
  File "/usr/local/lib/python3.5/dist-packages/s_tui/Sources/FreqSource.py", line 51, in read_msr
    e.message = e.message + "Unable to read file " + msr_file
AttributeError: 'OSError' object has no attribute 'message'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/s-tui", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 679, in main
    graph_controller = GraphController(args)
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 522, in __init__
    self.view = GraphView(self, args)
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 216, in __init__
    urwid.WidgetPlaceholder.__init__(self, self.main_window())
  File "/usr/local/lib/python3.5/dist-packages/s_tui/s_tui.py", line 456, in main_window
    freq_source = FreqSource(is_admin)
  File "/usr/local/lib/python3.5/dist-packages/s_tui/Sources/FreqSource.py", line 87, in __init__
    logging.debug(e.message)
AttributeError: 'AttributeError' object has no attribute 'message'
```

amanusk, 2018-02-17:
Thanks for opening the issue, 
Have you tried upgrading to the latest version, perhaps installing with pip3 will fix this. 
``` sudo pip3 install s-tui --upgrade```
Also, it seems the bug has to do with the `msr` module. Try running ```sudo modprobe msr``` before running s-tui. (When running with `sudo` the max turbo frequency is read from a file only available to root users).

gsedej, 2018-02-17:
I did update using pip3 and --upgrade. Also msr is loaded. Same happens with both kernel 4.15 and 4.16. 

With python2.7 it does not fail.
(Changing `#!/usr/bin/python3` to `#!/usr/bin/python2.7` in `/usr/local/bin/s-tui`)
But I don't see difference when using sudo (python2.7). Freq numbers are the same (normal/stress)

I think the "msr" is working, since there is some app in "go" to set some things.
```
~/git/go/bin$ sudo ./go-zenstates -l
Enabled - FID = 90 - DID = 8 - VID = 1C - Ratio = 36.00 - vCore = 1.37500
Enabled - FID = 80 - DID = 8 - VID = 2E - Ratio = 32.00 - vCore = 1.26250
Enabled - FID = 84 - DID = C - VID = 6A - Ratio = 22.00 - vCore = 0.88750
Disabled
Disabled
Disabled
Disabled
Disabled

```

amanusk, 2018-02-18:
Ok, The problem with python3 is caused because `message` was deprecated. This is an easy fix. 
The bigger problem is that the frequency is not registered correctly. 
Could you please tell me if the file `/dev/cpu/0/msr` exists after you run `sudo modprobe msr`, perhaps this was moved in newer kernels or in only available on Intel CPUs (or at a different location for AMD).

amanusk, 2018-02-18:
Looking further into it, maybe only the MSR that needs reading in different on AMD systems. 
Can you run `sudo cpupower frequency-info` and post the output, this should give max turbo per active cores

gsedej, 2018-02-18:
the files msr files exists. unfortunately i can't run `cpupower`, since it require package `linux-tools` the 4.15 and 4.16 kernels i am using form "ubuntu kernel ppa" web page doesn't have "linux-tools" packages for it's kernels. I am not familiar with building kernel on my own. I can use `cpufreq-info`

Btw, what does `s-tui` name stands for? I have trouble remembering the name.

Here are some relevant info:
```
$ ls /dev/cpu/*/msr -la
crw------- 1 root root 202,  0 feb 17 19:18 /dev/cpu/0/msr
crw------- 1 root root 202, 10 feb 18 20:55 /dev/cpu/10/msr
crw------- 1 root root 202, 11 feb 18 20:55 /dev/cpu/11/msr
crw------- 1 root root 202,  1 feb 18 20:55 /dev/cpu/1/msr
crw------- 1 root root 202,  2 feb 18 20:55 /dev/cpu/2/msr
crw------- 1 root root 202,  3 feb 18 20:55 /dev/cpu/3/msr
crw------- 1 root root 202,  4 feb 18 20:55 /dev/cpu/4/msr
crw------- 1 root root 202,  5 feb 18 20:55 /dev/cpu/5/msr
crw------- 1 root root 202,  6 feb 18 20:55 /dev/cpu/6/msr
crw------- 1 root root 202,  7 feb 18 20:55 /dev/cpu/7/msr
crw------- 1 root root 202,  8 feb 18 20:55 /dev/cpu/8/msr
crw------- 1 root root 202,  9 feb 18 20:55 /dev/cpu/9/msr

$ modinfo msr
filename:       /lib/modules/4.16.0-041600rc1-generic/kernel/arch/x86/kernel/msr.ko
license:        GPL
description:    x86 generic MSR driver
author:         H. Peter Anvin <hpa@zytor.com>
srcversion:     C3460BA1BECC23807FC76A0
depends:        
retpoline:      Y
intree:         Y
name:           msr
vermagic:       4.16.0-041600rc1-generic SMP mod_unload 

$ sudo cpufreq-info 
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  driver: acpi-cpufreq
  CPUs which run at the same hardware frequency: 0
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency: 4294.55 ms.
  hardware limits: 2.20 GHz - 3.60 GHz
  available frequency steps: 3.60 GHz, 3.20 GHz, 2.20 GHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance, schedutil
  current policy: frequency should be within 2.20 GHz and 3.60 GHz.
                  The governor "ondemand" may decide which speed to use
                  within this range.
  current CPU frequency is 2.20 GHz (asserted by call to hardware).
  cpufreq stats: 3.60 GHz:0,81%, 3.20 GHz:2,19%, 2.20 GHz:97,00%  (200809)
analyzing CPU 1:
  driver: acpi-cpufreq
  CPUs which run at the same hardware frequency: 1
  CPUs which need to have their frequency coordinated by software: 1
  maximum transition latency: 4294.55 ms.
  hardware limits: 2.20 GHz - 3.60 GHz
  available frequency steps: 3.60 GHz, 3.20 GHz, 2.20 GHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance, schedutil
  current policy: frequency should be within 2.20 GHz and 3.60 GHz.
                  The governor "ondemand" may decide which speed to use
                  within this range.
  current CPU frequency is 2.20 GHz (asserted by call to hardware).
  cpufreq stats: 3.60 GHz:0,63%, 3.20 GHz:1,21%, 2.20 GHz:98,16%  (86362)
analyzing CPU 2:
  driver: acpi-cpufreq
  CPUs which run at the same hardware frequency: 2
  CPUs which need to have their frequency coordinated by software: 2
  maximum transition latency: 4294.55 ms.
  hardware limits: 2.20 GHz - 3.60 GHz
  available frequency steps: 3.60 GHz, 3.20 GHz, 2.20 GHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance, schedutil
  current policy: frequency should be within 2.20 GHz and 3.60 GHz.
[... 98 lines cut ...]
  CPUs which need to have their frequency coordinated by software: 10
  maximum transition latency: 4294.55 ms.
  hardware limits: 2.20 GHz - 3.60 GHz
  available frequency steps: 3.60 GHz, 3.20 GHz, 2.20 GHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance, schedutil
  current policy: frequency should be within 2.20 GHz and 3.60 GHz.
                  The governor "ondemand" may decide which speed to use
                  within this range.
  current CPU frequency is 2.20 GHz (asserted by call to hardware).
  cpufreq stats: 3.60 GHz:1,06%, 3.20 GHz:2,60%, 2.20 GHz:96,34%  (208651)
analyzing CPU 11:
  driver: acpi-cpufreq
  CPUs which run at the same hardware frequency: 11
  CPUs which need to have their frequency coordinated by software: 11
  maximum transition latency: 4294.55 ms.
  hardware limits: 2.20 GHz - 3.60 GHz
  available frequency steps: 3.60 GHz, 3.20 GHz, 2.20 GHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance, schedutil
  current policy: frequency should be within 2.20 GHz and 3.60 GHz.
                  The governor "ondemand" may decide which speed to use
                  within this range.
  current CPU frequency is 2.20 GHz (asserted by call to hardware).
  cpufreq stats: 3.60 GHz:0,97%, 3.20 GHz:2,17%, 2.20 GHz:96,86%  (94408)

```

amanusk, 2018-02-18:
I see, If you run the stress option with s-tui (using pyhon2) is the recorded frequency `Cur freq` lower than the frequency reported by `Top freq`? I am less familiar with Turbo frequency options in AMD, and do not currently have AMD hardware to test this myself. 

(s-tui (pronounced Stewie)  is "stress terminal UI", but we just wanted a friendly name)

gsedej, 2018-02-18:
So the `Cur freq` has more "interesting" readings than `Top core 1 Freq` that stays 3600MHz all the time.

The `Cur freq` does seem to go up to 3692 MHz while in stress, else it's around 2GHz (e.g.: 1998, 2030, 2122, 1965, ...). The 3692 is probably all-core turbo (XFR/precision boost, or how do they call it)
Which I am really happy to finally able read actual clock. 
But when I set s-tui to Regular operation AND run `burnP6` or `burnK7` (single core stress) the Cur Freq stays around 2GHz, which is not expected - I guess that 1 core would go as high as possible (regarding cooling/power situation - i have AB350m and good air cooler)

Note, the Cur freq reading also works without sudo (python2.7)

Thanks for name explanation, i will be able to remember now!

amanusk, 2018-08-28:
@gsedej is there still an issue the requires addressing here?

gsedej, 2018-08-28:
The issue is resolved.

closed 2018-08-28 by gsedej