amanusk/s-tui #55: only laptop supported?

chensterliu, 2018-03-01:
I run s-tui on my desktop (i5 6500, Gentoo). Everything is fine except that the power value is N/A. I also tried it on my friend's Thinkpad laptop (CPU unknown, Arch). The power value is provided. I know some power monitor tool like powerstat, only supports battery charged device. Is the desktop not supported by s-tui?

By the way, I find another tool called Intel power gadget. I monitor the real-time power using its window version on the same desktop. This means there indeed exists the power sensor in my desktop CPU. Am I correct?

If s-tui also supports desktop, are there some tips to fix this problem?

Thanks

amanusk, 2018-03-01:
Hi,  
There should be no difference between laptop/desktop, s-tui does not use the battery metrics similar to `powertop` but rather direct statistics of the CPU. I actually have a system with i5 6500 showing power just fine (running Ubuntu  4.13).  

This makes me think the problem might be OS/kernel related. 
Do you have a directory `/sys/class/powercap/intel-rapl/` and read permissions for it?

chensterliu, 2018-03-01:
thanks for your relay. you're right, the `powercap` and `intel_rapl` are disabled in my kernel.
Problem solved.

BTW: is it possible to monitor the power of each core?

amanusk, 2018-03-01:
Unfortunately, I am not aware of a way to monitor power per core.  

In general, s-tui is showing average statistics for everything, but a per core granularity might be an option in the future.

amanusk, 2018-03-02:
I'll close this as the problem was solved.

closed 2018-03-02 by amanusk