amanusk/s-tui #56: Bad hang then stalls and weirdness

CottonEaster, 2018-03-28:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel i7-855OU, X1 Carbon G6
  * OS version: Linux Mint 18.3 Cinnamon
  * s-tui version: Whatever the pip install gives you
  * Installation method(pip/PPA/source code): pip install
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):

First test: after running for a few seconds, and possibly dependent upon user keypresses, the program hung the whole system hard, such that I have to Raise Elephants. I don't think I could have run the program's stress test tool - by accident - because that tool requires separate installation. I note that when I installed the program I did not install the `python-dev` package, though I later discovered that I lacked the package and installed that package. 

Second test - after installing python-dev - the program would, every few minutes, stall noticeably, but then recover and the rest of the system seemed unaffected, except that, even after I'd closed the program, the top of my screen would flicker occasionally in an alarming way. Relatedly: you should provide _un_install instructions.
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2018-03-28 01:49:56,879 [main()] [INFO ]  Started without root permissions
2018-03-28 01:49:56,880 [__init__()] [DEBUG]  stress is not installed
2018-03-28 01:49:56,881 [__init__()] [DEBUG]  stress-ng is not installed
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Sensor Label
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Sensor Label
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Package id 0
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Sensor Label
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Core 0
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Sensor Label
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Core 1
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Sensor Label
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Core 2
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Sensor Label
2018-03-28 01:49:56,885 [__init__()] [DEBUG]  Core 3
2018-03-28 01:49:56,886 [__init__()] [DEBUG]  Sensor Label
2018-03-28 01:49:56,886 [__init__()] [DEBUG]  
2018-03-28 01:49:56,886 [__init__()] [INFO ]  num cpus 8
2018-03-28 01:49:57,054 [update()] [INFO ]  Utilization recorded 3.9
2018-03-28 01:49:57,058 [get_power_usage()] [INFO ]  current 859961909.0 last 859961909.0
2018-03-28 01:49:57,059 [get_power_usage()] [INFO ]  Joule_Used 0.0 seconds_passed 0.00017023086547851562
2018-03-28 01:49:57,061 [update()] [INFO ]  Fan speed recorded0.0
2018-03-28 01:49:57,070 [update()] [INFO ]  Utilization recorded 8.3
2018-03-28 01:49:57,072 [get_power_usage()] [INFO ]  current 860006709.0 last 859961909.0
2018-03-28 01:49:57,072 [get_power_usage()] [INFO ]  Joule_Used 0.0448 seconds_passed 0.01375126838684082
2018-03-28 01:49:57,072 [get_power_usage()] [INFO ]  Max power updated 4
2018-03-28 01:49:57,073 [update()] [INFO ]  Fan speed recorded0.0
2018-03-28 01:49:57,073 [update_displayed_graph_data()] [INFO ]  Reading 1697
2018-03-28 01:49:57,073 [update_displayed_graph_data()] [INFO ]  Reading 8.3
2018-03-28 01:49:57,073 [update_displayed_graph_data()] [INFO ]  Reading 53.0
2018-03-28 01:49:57,074 [update_displayed_graph_data()] [INFO ]  Reading 3.257881290635782
2018-03-28 01:49:57,076 [update()] [INFO ]  Utilization recorded 16.7
2018-03-28 01:49:57,077 [get_power_usage()] [INFO ]  current 860035456.0 last 860006709.0
2018-03-28 01:49:57,077 [get_power_usage()] [INFO ]  Joule_Used 0.028747 seconds_passed 0.005048513412475586
2018-03-28 01:49:57,077 [get_power_usage()] [INFO ]  Max power updated 6
2018-03-28 01:49:57,078 [update()] [INFO ]  Fan speed recorded0.0
2018-03-28 01:49:57,078 [update_displayed_graph_data()] [INFO ]  Reading 2356
2018-03-28 01:49:57,078 [update_displayed_graph_data()] [INFO ]  Reading 16.7
2018-03-28 01:49:57,078 [update_displayed_graph_data()] [INFO ]  Reading 53.0
2018-03-28 01:49:57,079 [update_displayed_graph_data()] [INFO ]  Reading 5.694151456340024
2018-03-28 01:49:58,248 [update()] [INFO ]  Utilization recorded 1.2
2018-03-28 01:49:58,253 [get_power_usage()] [INFO ]  current 861493461.0 last 860035456.0
2018-03-28 01:49:58,254 [get_power_usage()] [INFO ]  Joule_Used 1.458005 seconds_passed 1.176093339920044
2018-03-28 01:49:58,255 [update()] [INFO ]  Fan speed recorded0.0
2018-03-28 01:49:58,255 [update_displayed_graph_data()] [INFO ]  Reading 2110
2018-03-28 01:49:58,255 [update_displayed_graph_data()] [INFO ]  Reading 1.2
2018-03-28 01:49:58,256 [update_displayed_graph_data()] [INFO ]  Reading 37.0
2018-03-28 01:49:58,256 [update_displayed_graph_data()] [INFO ]  Reading 1.2397017740948364
2018-03-28 01:49:59,419 [update()] [INFO ]  Utilization recorded 2.3
2018-03-28 01:49:59,425 [get_power_usage()] [INFO ]  current 863293383.0 last 861493461.0
2018-03-28 01:49:59,425 [get_power_usage()] [INFO ]  Joule_Used 1.799922 seconds_passed 1.1716148853302002
2018-03-28 01:49:59,427 [update()] [INFO ]  Fan speed recorded0.0
[... 904 lines cut ...]
2018-03-28 01:53:42,959 [update()] [INFO ]  Fan speed recorded0.0
2018-03-28 01:53:42,959 [update_displayed_graph_data()] [INFO ]  Reading 1751
2018-03-28 01:53:42,960 [update_displayed_graph_data()] [INFO ]  Reading 3.3
2018-03-28 01:53:42,961 [update_displayed_graph_data()] [INFO ]  Reading 41.0
2018-03-28 01:53:42,962 [update_displayed_graph_data()] [INFO ]  Reading 1.5734233833250588
2018-03-28 01:53:44,135 [update()] [INFO ]  Utilization recorded 4.6
2018-03-28 01:53:44,141 [get_power_usage()] [INFO ]  current 1454999632.0 last 1453673463.0
2018-03-28 01:53:44,141 [get_power_usage()] [INFO ]  Joule_Used 1.326169 seconds_passed 1.1832456588745117
2018-03-28 01:53:44,142 [update()] [INFO ]  Fan speed recorded0.0
2018-03-28 01:53:44,142 [update_displayed_graph_data()] [INFO ]  Reading 1392
2018-03-28 01:53:44,143 [update_displayed_graph_data()] [INFO ]  Reading 4.6
2018-03-28 01:53:44,144 [update_displayed_graph_data()] [INFO ]  Reading 37.0
2018-03-28 01:53:44,145 [update_displayed_graph_data()] [INFO ]  Reading 1.1207892376815776
2018-03-28 01:53:44,671 [unhandled_input()] [DEBUG]  Caught q
2018-03-28 01:53:44,671 [unhandled_input()] [DEBUG]  None
2018-03-28 01:53:44,671 [kill_child_processes()] [DEBUG]  No such process

  
### Step 3: Reproduce the problem:
#### Steps to reproduce:

Unknown. Seemingly simply install and run on my machine.
```
* Replace this with relevant code to reproduce the problem *
```

amanusk, 2018-04-04:
Thank you for the detailed bug report, 
Unfortunately I have not had a chance to experience anything similar to you description. In part, the python-dev package should not affect on relatively new python versions.  
My only theory at the moment is that the issue might have something to do with the terminal emulator used to run s-tui. 

I'll keep this open to monitor similar behaviour on other systems.

CottonEaster, 2018-07-10:
I am happy enough to close this issue for, in a way, it has been solved. It has been solved in this way/sense: on Mint 19, the program seems to work perfectly.

amanusk, 2018-07-10:
Thanks for the update. 

I was suspecting it was somehow related to Mint 18.3 when I came across a reddit thread on /r/linuxmint were a similar behavior was mentioned regarding some other program (cannot find it now). 
Also, is seems there was no other distribution exhibiting this behavior. 

Indeed there was no fix, but a least we have identified *a* reason for this issue.

closed 2018-07-10 by CottonEaster