amanusk/s-tui #57: Please support FreeBSD for Frequency/Temperature/Power

yurivict, 2018-04-14:
Currently it says ```N/A```.

Version 0.6.7.4 on FreeBSD 11.1

amanusk, 2018-04-23:
Unfortunately I currently do not have a FreeBSD system I can test this on. 
Anyone who is willing to add support is welcome to issue a PR.

amanusk, 2018-09-24:
@yurivict I tried adding some functionality for s-tui to work better with FreeBSD. 
The work is available on branch: https://github.com/amanusk/s-tui/tree/free_bsd

Could you help testing this?

This is a mitigation of the lack of this functionality in `psutil`, but hopefully, I can work on it :)

yurivict, 2018-09-24:
@amanusk  Thank you for working on the FreeBSD support!
It works.
I've updated the port. https://www.freshports.org/sysutils/s-tui/

GrandAdmiralThrawn, 2019-12-05:
I tried to run 1.0.0-b2 on FreeBSD UNIX today (because of its multicore/-thread visualization support), and it appears to work rather well, at least on x86_64. I ported the patch for get_processor_name() over to the new helper_functions.py and now it manages to fetch the real CPU name from sysctl as well, instead of just displaying "amd64".

As for the other changes... I don't really speak Python, so I didn't manage to port them over, but it would be cool if we could get 1.0.0 for FreeBSD as well. :)

Thanks for your work!

amanusk, 2019-12-06:
That's great to know!
The work to add functionality to FreeBSD has mostly been done in `psutil`, so it can work with `s-tui` out of the box.
I was hoping off with releasing version 1.0 until bugs were reported and ironed out. I believe it is quite stable at this stage and will hopefully release it "officially" very soon.

GrandAdmiralThrawn, 2019-12-06:
It seems to work rather well indeed, even out-of-the-box on FreeBSD 12.1 (like: Multicore, load, and clock speed displays work, at least on my Intel hardware). The only thing I had to patch taking code from the FreeBSD branch was the CPU name detection in helper_functions.py. I wasn't aware that most of the FreeBSD-specific code had been moved into `psutil`.

I'm now getting one of those new Threadripper CPUs, and I'm going to use FreeBSD with that as well if everything works. And I'm planning to use `s-tui` for load monitoring with that wide CPU, since I like it a bit more than htop. We'll see how detailed that brand-new chip shows up in sysctl. ;)

yurivict, 2019-12-06:
> The only thing I had to patch taking code from the FreeBSD branch was the CPU name detection in helper_functions.py.

Please also submit your patch for the port https://bugs.freebsd.org/bugzilla/ so that other users can benefir from it too.

GrandAdmiralThrawn, 2019-12-06:
Hello yurivict,

I'm not sure if that makes sense. From what I got, the s-tui version in the FreeBSD ports tree is 0.8.3 from the free_bsd branch of s-tui? That one is already patched. Because I took the code parts from that version to make 1.0.0b work in the same way as well.

Maybe amanusk could update helper_functions.py, so that version 1.0.0 displays CPU names properly on FreeBSD out-of-the-box as soon as it's being released officially?

Besides, I've never submitted a patch anywhere, would need to look into how to generate a patch properly etc., as I'm not really a developer. ;)

amanusk, 2019-12-06:
@GrandAdmiralThrawn, I have added the cpu name fix in this branch https://github.com/amanusk/s-tui/tree/freebsd_fixes

Please tell me if this works for you (I don't currently have FreeBSD installed on an active system).

Also please update if any kernel modules were needed to make things work. Of course, you are welcome to make pull requests with additions/fixes and updates to the README

GrandAdmiralThrawn, 2019-12-06:
I don't even know how a pull request works! :) Like I said, I'm no developer.

But I will try that branch on monday, when I have access to my FreeBSD notebook again. I'll let you know how it goes.

I'll also try to unload as many kernel modules as possible to see whether they affect the output of s-tui.

GrandAdmiralThrawn, 2019-12-09:
> @GrandAdmiralThrawn, I have added the cpu name fix in this branch https://github.com/amanusk/s-tui/tree/freebsd_fixes

I just checked out your new freebsd_fixes branch of 1.0.0b2 on FreeBSD 12.0, and it works out of the box and detects my CPU's name correctly!

Out of curiosity: Is there a reason as to why you have removed support for MacOS X / macOS? Because it suffers from the same issue with CPU name detection. It's practically the same as with FreeBSD. The following code for `get_processor_name()` in helper_functions.py makes it work again:

```
    elif platform.system() == "Darwin":
        return subprocess.check_output(['sysctl', '-n', 'machdep.cpu.brand_string']).strip()
```

I just tested this for an older MacOS X Sierra 10.12.5, and it works. Yeah, that's not a current version of the operating system, but it's all I have here right now when it comes to Apple stuff.

GrandAdmiralThrawn, 2019-12-21:
And one more: My new Threadripper 3970X machine is (mostly) complete and FreeBSD 12.1 is running. s-tui works, But I've found a cosmetic problem with having so many logical CPUs. ;) See here:

http://www.xin.at/thrawn/pics/screenshots/s-tui/s-tui-1.0b-FreeBSD-64cpus.png

My monitor just ain't big enough for this! ;) With so many CPUs, you can't get any meaningful per-cpu history anymore, because s-tui shows all CPUs in one row. I think it would be better to have a grid of columns and rows, like e.g. htop or Windows task manager do it.

Thanks!

amanusk, 2019-12-21:
First, it look pretty cool :). 
I have given it some thought, but honestly, don't have a clear idea how to do this. A grid makes sense, but then there are the other sensors as well, so a grid of grids?

I guess the solution for now would be to only have some of the shown. But this definitely does not give the same effect as seeing 64 cores.
(The number of cores in new CPUs is too damn high)..

GrandAdmiralThrawn, 2019-12-21:
A "grid of grids" may be too much for people to display on any kind of terminal, if there are 32 CPUs or more and if they choose to display all sensors individually. But it may still work if a user configures s-tui to display only what they need to see (in my case: load per core). One could display averages for the rest or simply disable what's not needed. Thankworthily, s-tui is already flexible enough for that.

Also: Yeah, core counts are pretty damn high by now, but I'm happy about it, because now I can crunch through all the work that has been piling up. ;)

And it's not like this is going to change... In Q1/2020, AMD is expected to release a 64-core / 128-thread version of the processor I'm using, and in the server field you can have 256 or even 512 logical CPUs per system.

Maybe grouping by NUMA node (=usually a socket) may be an idea. FreeBSD isn't ready for that yet I think, but Windows and Linux with libnuma should be. Not sure about macOS. And not sure whether you can analyze a system's NUMA topology from within Python. It's just an idea...

amanusk, 2020-01-22:
@yurivict, s-tui has been updated to version 1.0, which should have all the freeBSD specific features baked in (ported upstream to psutil).
Could you verify and update the port?