amanusk/s-tui #59: Add run time for stress test

GM-Script-Writer-62850, 2018-05-05:
When running a stress test it should have a 'wall time' as memtest86 calls it
It would be nice to have a clock showing how long the stress test has been running

amanusk, 2018-05-07:
Thank, 
That's a great idea for an addition.

amanusk, 2018-08-20:
@GM-Script-Writer-62850 I have implemented a timer in branch `clock`  (PR #75)
https://github.com/amanusk/s-tui/tree/clock
Would you care to test it?

GM-Script-Writer-62850, 2018-08-20:
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/chad/Desktop/s-tui-clock/s_tui/s_tui.py", line 973, in <module>
    main()
  File "/home/chad/Desktop/s-tui-clock/s_tui/s_tui.py", line 904, in main
    graph_controller.main()
  File "/home/chad/Desktop/s-tui-clock/s_tui/s_tui.py", line 762, in main
    self.main()
  File "/home/chad/Desktop/s-tui-clock/s_tui/s_tui.py", line 754, in main
    self.animate_graph()
  File "/home/chad/Desktop/s-tui-clock/s_tui/s_tui.py", line 771, in animate_graph
    self.view.update_displayed_information()
  File "/home/chad/Desktop/s-tui-clock/s_tui/s_tui.py", line 252, in update_displayed_information
    self.controller.stress_time = (time.perf_counter() -
AttributeError: 'module' object has no attribute 'perf_counter'
```

amanusk, 2018-08-22:
Thanks, I Forgot to test Python2 compatibility, should be working now. (In master).

GM-Script-Writer-62850, 2018-08-22:
Looks like it works, wish I had that a week ago since i now have a stable OC and do not need to stress test now...

Looks like you are starting to run low on vertical space, so here are a couple ideas

What if the timer was intended like a sub bullet point under stress?
What if it was in parentheses next to stress?

amanusk, 2018-08-22:
The vertical space does not really run out, you can always scroll to see the rest (or decrease the font), but it is nice to have everything in one window. 
I'll think about how to change it around a bit, maybe only have the timer optional with a flag or maybe just make it take less space.

GM-Script-Writer-62850, 2018-08-22:
Wait it can scroll!?
You could also merge your 5 options into a single box
save settings can be moved to under control options
quit can get a flag to be replaced with the esc key, with optional confirm prompt
space below/above boxed items can be trimmed a bit

I am not out of space, but once you add per core data (freq, utilization, temperature) you are going to be very much out of space and fitting graph data will be a nightmare with a dual socket board running Epyc 7601 processors (64 cores 128 threads)

amanusk, 2018-08-22:
> Wait it can scroll!? 

Indeed :) You can also use the `G` or `PG-DN` to go to the bottom of the side bar. 

`Esc` ,`q` and `Ctrl+C` all work for exit, so perhaps the `quit` button is redundant. 

>You could also merge your 5 options into a single box 

Do you want to give it a try and issue a PR?

GM-Script-Writer-62850, 2018-08-22:
Esc does not work... the other do though
PR?

amanusk, 2018-08-22:
> Esc does not work... the other do though

Right, it is saved for exiting the sub-menus

GM-Script-Writer-62850, 2018-08-23:
Agreed

closed 2018-08-23 by GM-Script-Writer-62850