amanusk/s-tui #65: Two different installation problems

CottonEaster, 2018-06-18:
### Step 1: Describe your environment

  * Intel i7-8550U
  * OS version: Mint 18.3 Cinnamon
  * s-tui version: 0.6.7-1 and (see below) 0.7.6
  * Installation method(pip/PPA/source code): PPA in first case, pip in second
  
### Step 2: Describe the problem:
#### Observed Results:

When I installed via PPA, this happened when I ran the program:

```
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3086, in <module>
    @_call_aside
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3070, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3099, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 576, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 589, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 778, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'psutil>=5.2.0' distribution was not found and is required by s-tui
```
Mint has python-psutil version 3.4.2.1 in its repositories.

When using the pip install, this happened:

```
sudo pip install s-tui
The directory '/home/<user>/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/<user>/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting s-tui
  Downloading https://files.pythonhosted.org/packages/b3/bf/aeb38a3fc3f84ae7261f2d24639b6f84081a3c037fae2a2db33715093375/s-tui-0.7.6.tar.gz
Requirement already satisfied: urwid>=1.3.1 in /usr/lib/python2.7/dist-packages (from s-tui)
Collecting psutil>=5.2.0 (from s-tui)
  Downloading https://files.pythonhosted.org/packages/51/9e/0f8f5423ce28c9109807024f7bdde776ed0b1161de20b408875de7e030c3/psutil-5.4.6.tar.gz (418kB)
    100% |████████████████████████████████| 419kB 2.8MB/s 
Installing collected packages: psutil, s-tui
  Found existing installation: psutil 3.4.2
    DEPRECATION: Uninstalling a distutils installed project (psutil) has been deprecated and will be removed in a future version. This is due to the fact that uninstalling a distutils project will only partially uninstall the project.
    Uninstalling psutil-3.4.2:
      Successfully uninstalled psutil-3.4.2
  Running setup.py install for psutil ... done
  Running setup.py install for s-tui ... done
Successfully installed psutil-5.4.6 s-tui-0.7.6
You are using pip version 9.0.3, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
$ s-tui
bash: /usr/bin/s-tui: No such file or directory
$ s-tui
python: can't open file 's-tui': [Errno 2] No such file or directory
```

PS: This is fifth time I have tried to get s-tui working on Mint.

amanusk, 2018-06-18:
Hi, 
Thanks for the bug report. 

I believe there is a collision between the two installations. 
Could you please try one of the following:
If you want to install via PPA:
Use `sudo apt install python3-s-tui`, and update `psutil` via `pip install psutil --upgrade`. A newer version of `psutil` is required for all the features to work. 

If you want to install via pip (which is usually the newest version):
First try completely uninstalling any installation of s-tui that was installed via PPA:
`sudo apt remove --purge python*-s-tui`
Possibly psutil as well:
`sudo apt remove --purge python*-psutil`
(The * if for both python2 and python3 versions) 
Try removing `/usr/bin/s-tui` and `/usr/local/bin/s-tui` if they were created and not removed by `apt remove`. 
Then install s-tui with `sudo pip install s-tui`. This will install the latest versions of `psutil` and `urwid` which are the only dependencies of `s-tui`.
Now try running `s-tui` in a new shell. 

Sorry if this sounds a bit complected, this is caused because the system is confused between the two installation methods and  the fact that only an older version of `psutil` is available in 16.04 based systems. Newer version (18.04) should have all the needed prerequisites. 

Please update if this worked out for you.

CottonEaster, 2018-06-18:
Thanks. I used the PPA method that you described - preceeded by `sudo apt remove --purge python*-s-tui` (but not `sudo apt remove --purge python*-psutil`, because `apt` told me it would take half my system with it) and s-tui then ran. However, and as I've reported in a previous, separate bug report, s-tui then stalled and indeed temporarily hung my whole machine.

amanusk, 2018-06-18:
I see, 
I'll try to reproduce and investigate, what terminal emulator do you use?

CottonEaster, 2018-06-18:
xfce4-terminal

Thanks. My other bug report is [here](https://github.com/amanusk/s-tui/issues/56).

amanusk, 2018-06-18:
I have tried running Mint 18.3 Cinnamon with xfce4-terminal from live USB and unfortunately was not able to reproduce a hang. 
If you have the chance, and want to help debugging the issue, could you please try connecting to the system over SSH, and see if just running s-tui from a remote session causes a hang? (This might mean the issue is related to the newest gen CPUs)
If there is no effect, the issue could be related to the graphical interface of Mint. 
(If you are able to run a different distro on the same system, it will could help pinpoint the cause)

CottonEaster, 2018-06-18:
I'd be happy to take the SSH approach - in a few days to a week's time - if you'll point me towards a primer on how to do that. Such things - SSH, FTP, Putty - aren't things I've much experience with (and also I think I've rather locked my system down against them, so some re-enabling will be needed).

amanusk, 2018-06-20:
I'll by happy to try and explain. 
I have recently opened a gitter room for s-tui https://gitter.im/s-tui/Lobby
We can discuss it over there.

amanusk, 2018-06-23:
I'll close this as the installation issue has been solved. 
Please update in your previous issue on hangs #56.

closed 2018-06-23 by amanusk