amanusk/s-tui #68: Negative wattage reading

GM-Script-Writer-62850, 2018-08-05:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): _Intel_
  * OS version: xubuntu 18.04
  * s-tui version: 0.7.4
  * Installation method(pip/PPA/source code): pip
  
Motherboard: Asrock Z97 Killer
CPU: Intel i5-4690k (overclocked of course, around 4.3Ghz, vcc; 1.225)
Cooler: NH-D15

### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):
Oddity in power monitor
at around 45min into a stress test (3 times in a row now)
it shows a single odd reading, it is a 5 or 6 digit negative wattage value

GM-Script-Writer-62850, 2018-08-09:
if you think the -d option will give you some useful data i can try it this weekend when i try to get another 100 Mhz in my overclock

amanusk, 2018-08-18:
It seems the problem occurs when the file we are reading for wattage usage elapses. 
We are actually reading power consumption in joules, and dividing by time to get Watts.

I have added a simple fix, to keep the previous reading when this happens. Could you please try cloning the `master` branch and test if this fixes the issue for you?

GM-Script-Writer-62850, 2018-08-18:
I'll give it a try when i try to get another 100 Mhz later today that I did not get a chance to last week.

GM-Script-Writer-62850, 2018-08-19:
i have done 2 long stress test and did not notice any negative readings appear, i could have not been looking at the screen at the time, but so far so good

as for the 100 Mhz, i did not like the voltage required... so +050 Mhz

amanusk, 2018-08-19:
Thanks, 
You can use `--csv` to record all values into a file, so you can see all values in retrospect

amanusk, 2018-08-27:
@GM-Script-Writer-62850 did you get a chance to see if this was fixed?
If so, could this be closed?

GM-Script-Writer-62850, 2018-08-27:
i have not used --csv
i only checked for it that one time

GM-Script-Writer-62850, 2018-08-27:
If the system crashes during a stress test will any data observed be saved?
* I do not care if the data is missing a closing bracket or something, i can fix that

amanusk, 2018-08-28:
The data is saved to the CSV in this format:

| Time | Top Freq | Cur Freq | Perf Lost | Utilization | Max Temp | Cur Temp | Max Power | Cur Power | Fan |
|------|-----------|-----------|----------|-------------|------------|-------------|------------|-------------|-----|
| 2018-08-28_08:22:05 | 3300 MHz | 2275.0 MHz | 0 (N/A) run sudo | 28.6 % | 55.0 C | 55.0 C | 9.0 W | 8.2 W | 3080.0 RPM |

It is written each time a sample is takes, so if the system crashes, the file will be kept

amanusk, 2018-11-30:
Assuming this is closed. Please reopen if needed.

closed 2018-11-30 by amanusk