amanusk/s-tui #70: Fan speed, needs support for more fans

GM-Script-Writer-62850, 2018-08-09:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: xubuntu 18.04
  * s-tui version: 0.7.4
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results:
Only shows one fan speed
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
* Replace this with debug output *
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
Have more than one fan rpm input
```
~$ sensors|grep RPM
Chassis Fan 1 Speed:       0 RPM  (min =    0 RPM)
CPU Fan 1 Speed:         383 RPM  (min =    0 RPM)
Chassis Fan 3 Speed:     415 RPM  (min =    0 RPM)
Chassis Fan 4 Speed:     457 RPM  (min =    0 RPM)
Power Fan Speed:           0 RPM  (min =    0 RPM)
CPU Fan 2 Speed:         451 RPM
```
At the very lease it would be nice to select a fan to monitor instead of getting the 1st fan detected

amanusk, 2018-08-15:
You can set a custom fan using the CLI with `s-tui -cf <fan>` , run  `s-tui --help` to see how to configure it. 
Regardless, adding all possible fans in the bottom of the side menu could be a good idea.
I'll try experimenting with it

amanusk, 2019-04-01:
@GM-Script-Writer-62850
Master branch was updated to a new version, it should now show all available fans in the stats.

Help testing is appreciated

GM-Script-Writer-62850, 2019-04-01:
*installs update*
holy... SOOO many charts... wonder how it looks with hyper-threading...
[image]
Be nice if RPM would fit on the same line, thought i think you could just drop RPM from the speed line and put it in parenthesis in the section title

There should be a option to assign custom aliases to sensors like i can in lm-sensors.conf

wordlessly, 2019-04-08:
I don't get ANY fan speed values at all. I have a NUC8i5BEH with a single CPU-fan. Any idea how to fix this issue and get fan speeds?

amanusk, 2019-04-08:
@wordlessly, do you get any results with `sensors`?

GM-Script-Writer-62850, 2019-04-08:
Do not forget to run `sensors-detect`

wordlessly, 2019-04-17:
Below it the output of sensors AND sensors-detect. 

I think I did run sensors-detect before but I am not sure if I added the suggested lines to /etc/modules. Looking at /etc/modules there is nothing in there. Still, calling "sensors" gives some output. How is this possible? Should I modify my /etc/modules as suggested by sensors-detect? Would it then show my fan speed?

`sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +53.0°C  (high = +100.0°C, crit = +100.0°C)

acpitz-virtual-0
Adapter: Virtual device
temp1:       -263.2°C  
temp2:        +27.8°C  (crit = +119.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +54.0°C  

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +54.0°C  
`

`sudo sensors-detect
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Intel(R) Client Systems NUC8i5BEH [J72747-302]
# Board: Intel Corporation NUC8BEB
# Kernel: 4.15.0-47-generic x86_64
# Processor: Intel(R) Core(TM) i5-8259U CPU @ 2.30GHz (6/142/10)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8987

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:9da3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: i915 gmbus dpb (i2c-0)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpc (i2c-1)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus misc (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpd (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: DPDDC-B (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: DPDDC-C (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

To load everything that is needed, add this to /etc/modules:
#----cut here----
# Chip drivers
coretemp
#----cut here----
If you have some drivers built into your kernel, the list above will
contain too many modules. Skip the appropriate ones!

Do you want to add these lines automatically to /etc/modules? (yes/NO)

Unloading cpuid... OK
`

GM-Script-Writer-62850, 2019-04-17:
For my board i asked on the ubunutforums and a person suggested running this command
`sudo modprobe nct6775` no idea if that will work for you
you can see a list of different models by running this command
`/lib/modules/$(uname -r)/kernel/drivers/hwmon/`
you could try modprobing every one and checking the output of sensors to see which ones work if any

wordlessly, 2019-04-17:
It tells me: modprobe: ERROR: could not insert 'nct6775': No such device

Is executing 'sudo modprobe abc' dangerous?

GM-Script-Writer-62850, 2019-04-18:
as far as i know it is not, if the model will not help nothing should happen
if you are worried about it save any unsaved data and if something goes wrong a reboot will reset everything you did with random modprobes

I would probably start with these modules:
`ls /lib/modules/$(uname -r)/kernel/drivers/hwmon/|grep nct`
*I am using Ubuntu 18.04 (xfce desktop)

wordlessly, 2019-04-18:
From all 4 nct modules I was able to load those two:
modprobe nct7802
modprobe nct7904

Still, sensors-detect --auto detected no additional sensors.

Maybe my version of sensors is just too old? I am running basically Ubuntu 18.04 as well (actually Mint)

sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)

GM-Script-Writer-62850, 2019-04-18:
your platform (intel 8XXX CPU) is a few years newer than mine (intel 4XXX CPU) and iirc my platform was launched are using the time 18.04 was released so that is very likely, try loading the testing build of 19.04 and see if you can see/get any sensors on a live session
run sensors, sensors-detect tries to figure out what modules you need, running modprobe manually bypass using sensors-detect as we are assuming sensors-detect is not working
```
$ sensors -version
sensors version 3.4.0 with libsensors version 3.4.0
```

amanusk, 2019-04-22:
@wordlessly, did you manage to figure it out?

wordlessly, 2019-04-22:
No. I did not so far. I tested with the most recent distributions like Manjaro and Xubuntu 19.04. I did run sensors / sensors-detect with those distros but with no success. Same result as Linux Mint 19.1. No fan speed.

I didn't run s-tui though with those distris. Do you think s-tui could do on those distris what 'sensors' cannot?

amanusk, 2019-04-22:
> I didn't run s-tui though with those distris. Do you think s-tui could do on those distris what 'sensors' cannot?

Sadly probably not.
It seems there is no support for the fan sensors of NUCs in Linux.

wordlessly, 2019-04-23:
> > I didn't run s-tui though with those distris. Do you think s-tui could do on those distris what 'sensors' cannot?
> 
> Sadly probably not.
> It seems there is no support for the fan sensors of NUCs in Linux.

Thanks for the info. 

Is fan support missing only for the NUC8 or all NUCs? Linux is just ignoring the fan-sensor at the moment, right? The NUC-Bios seems able to measure fan-speed.

[image]

amanusk, 2019-04-23:
> Is fan support missing only for the NUC8 or all NUCs? Linux is just ignoring the fan-sensor at the moment, right? The NUC-Bios seems able to measure fan-speed

I am not 100% percent sure what exactly is missing. I tried accessing fan speed on a NUC (7 I think) without success, so my conclusions are base on empirical evidence :)

I have experienced cases where `sensors` did not detect fans and `psutil/s-tui` did, so it's worth a shot.

My guess is that there are no driver support in the Kernel.

closed 2020-07-11 by amanusk