amanusk/s-tui #71: Crash on first run on Ubuntu 18.04 install from apt, missing distutils dependency

mikehardy, 2018-08-10:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: Ubuntu 18.04
  * s-tui version: 0.7.6
  * Installation method(pip/PPA/source code): PPA
  
### Step 2: Describe the problem:
#### Observed Results:

  crash with Python traceback indicating it couldn't find distutils
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
  File "/usr/bin/s-tui", line 9, in <module>
    load_entry_point('s-tui==0.7.6', 'console_scripts', 's-tui')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 43, in <module>
    from distutils.spawn import find_executable
ModuleNotFoundError: No module named 'distutils.spawn'
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Install using PPA method on Ubuntu 18.04
2. run s-tui
3. see the callback

I think the package just needs a dependency on python3-distuils - that solved it for me:

```
sudo apt-get install python3-distutils
```

Worked great after that, thank you!, and easily helps visualize why I need to re-paste my factory-fresh Dell laptop CPU :-)

mikehardy, 2018-08-10:
Note that this may have been a move from package name "python3-distutils-extra" (pre-18.04) to "python3-distutils" (in 18.04). It appears that 18.04 still has python3-distutils-extra, which depends on python3-distutils, so adding a python3-distutils-extra dependency may be the simplest fix that covers people from 14.04LTS and up

You can see the 18.04 package here as well as the related previous-release package hits below it (and if you follow them you see a depends on distutils for 18.04, while that package doesn't seem to exist in previous distributions but the -extra package provides the Python code I believe)
https://packages.ubuntu.com/search?keywords=python3-distutils

amanusk, 2018-08-20:
Thanks for the detailed issue report. 
I tried to install from PPA on 18.04 running off a live USB and was unable to recreate the issue. 
Was this a minimal install / regular install?

mikehardy, 2018-08-21:
Hmm - this wasn't minimal but I'm trying to remember the options I chose on install. I still have the installation logs in /var/log/installer but I'm not sure how to get whatever information would help nail this down. If you can think of anything I could pull out of there let me know, otherwise I don't what to say except I took the ubuntu desktop iso, used etcher to put it on a USB key, then ran the installer with defaults.

amanusk, 2018-11-30:
Dependency on `distutils` removed. Closing.

closed 2018-11-30 by amanusk

shotor, 2020-03-09:
still getting this issue, am I doing something wrong?

amanusk, 2020-03-10:
@shotor, are you trying to install?

shotor, 2020-03-10:
Yes, well install succeeds. I get the same error message when trying to run
s-tui

amanusk, 2020-03-12:
Ok, but what did you use to install pip/apt? which OS?

shotor, 2020-03-12:
my bad, I'm using apt and ubuntu 18.04
I installed through the PPA

```sh
❯ s-tui --version
s-tui 0.7.6 - (C) 2017-2018 Alex Manuskin, Gil Tsuker
    Released under GNU GPLv2
```