amanusk/s-tui #72: Enhancement possibility: support graphics freq/power

mikehardy, 2018-08-10:
I noticed with the Intel power gadget they can do the built-in intel GPU information as well, would be neat if s-tui could also access it:
https://software.intel.com/en-us/articles/intel-power-gadget-20

Separately, the Intel power gadget doesn't handle it but if there was an option to handle discrete graphics (for laptops that have a discrete GPU) it would be extra cool if it could monitor those also

mikehardy, 2018-09-02:
conky can display - if configured and you have the right things installed - an example from here https://askubuntu.com/a/1020887/858118

#------------+
# Nvidia GPU |
#------------+
${color orange}${hr 1}${if_existing /tmp/nvidia}
${color2}${voffset 5}${execpi .001 (nvidia-smi --query-gpu=gpu_name --format=csv,noheader)} ${color}@ ${color green}${execpi .001 (nvidia-smi --query-gpu=clocks.sm --format=csv,noheader)} ${alignr}${color}Temp: ${color green}${execpi .001 (nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader)} C
${color}${voffset 5}Ver: ${color green}${execpi .001 (nvidia-smi --query-gpu=driver_version --format=csv,noheader)} ${color} P-State: ${color green}${execpi .001 (nvidia-smi --query-gpu=pstate --format=csv,noheader)} ${alignr}${color}BIOS: ${color green}${execpi .001 (nvidia-smi --query-gpu=vbios_version --format=csv,noheader)}
${color}${voffset 5}GPU:${color green}${execpi .001 (nvidia-smi --query-gpu=utilization.gpu --format=csv,noheader)} ${color}Ram:${color green}${execpi .001 (nvidia-smi --query-gpu=utilization.memory --format=csv,noheader)} ${color}Pwr:${color green}${execpi .001 (nvidia-smi --query-gpu=power.draw --format=csv,noheader)} ${alignr}${color}Freq: ${color green}${execpi .001 (nvidia-smi --query-gpu=clocks.mem --format=csv,noheader)}
#------------+
# Intel iGPU |
#------------+
${else}
${color2}${voffset 5}Intel® Skylake GT2 HD 530 iGPU @  ${color green}${execpi .001 (cat /sys/class/drm/card0/gt_cur_freq_mhz)} MHz
${color}${goto 13}Min. Freq:${goto 120}${color green}${execpi .001 (cat /sys/class/drm/card0/gt_min_freq_mhz)} MHz${color}${goto 210}Max. Freq:${alignr}${color green}${execpi .001 (cat /sys/class/drm/card0/gt_max_freq_mhz)} MHz
${color orange}${hr 1}

mclang, 2018-10-03:
This would be really great addition and maybe not so hard to implement using conky example.

amanusk, 2018-10-03:
This is definitely on the road map. 
The next milestone is to enable per-core view of the metrics. This will lead to many changes in the current code base. Once it is implemented, it should be easier to add support for a graphics card, and hopefully, multiple graphics cards as well.

mikehardy, 2020-08-09:
Closing out old issues / personal git cleaning...

closed 2020-08-09 by mikehardy