amanusk/s-tui #83: Feature request: display Power/Thermal/Current throttling

sanderboom, 2018-09-01:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel Core i7-8550U
  * OS version: 4.18.4-1-MANJARO x86_64
  * s-tui version: 0.8.2-1
  * Installation method(pip/PPA/source code): `python2 setup.py install`
  
First of all: thanks for the great software.

Secondly: IMO it would be a very valuable addition if `s-tui` could display the onset/offset of Thermal/Power/Current throttling. 

Thanks for taking a look at this.

amanusk, 2018-09-03:
@sanderboom ?

sanderboom, 2018-09-04:
@amanusk Thanks for the reply. For a screenshot of XTU I had to boot windows - Something I typically try to avoid ;-)

The goal of these statistics is to indicate if/when the different types of throttling are active. 

The first screenshot: XTU showing active Power Limit Throttling (the white line) during 2 stress tests.
[image]

The second screenshot: XTU showing active Thermal Throttling (the green line) for a while during the last stress test (triggered by raising the TDP's), followed by the Power Limit Throttling kicking in.
[image]

Thanks for looking into this.

amanusk, 2018-09-04:
> For a screenshot of XTU I had to boot windows - Something I typically try to avoid ;-)

I feel your pain ;)

Do you have a source/reference of how could this be monitored in Linux? Are these events written to a certain MSR? Does this show up in `dmesg`? or some other log file?

sanderboom, 2018-09-04:
I've seen the following a bit too often (via `journalctl -g "Package temperature"`): 

```
jul 25 11:55:49 T480s kernel: CPU1: Package temperature above threshold, cpu clock throttled (total events = 28)
jul 25 11:55:49 T480s kernel: CPU2: Package temperature above threshold, cpu clock throttled (total events = 28)
jul 25 11:55:49 T480s kernel: CPU5: Package temperature above threshold, cpu clock throttled (total events = 28)
jul 25 11:55:49 T480s kernel: CPU7: Package temperature above threshold, cpu clock throttled (total events = 28)
jul 25 11:55:49 T480s kernel: CPU6: Package temperature above threshold, cpu clock throttled (total events = 28)
jul 25 11:55:49 T480s kernel: CPU3: Package temperature above threshold, cpu clock throttled (total events = 28)
jul 25 11:55:49 T480s kernel: CPU4: Package temperature above threshold, cpu clock throttled (total events = 28)
jul 25 11:55:49 T480s kernel: CPU0: Package temperature above threshold, cpu clock throttled (total events = 28)
jul 25 11:55:49 T480s kernel: CPU5: Package temperature/speed normal
jul 25 11:55:49 T480s kernel: CPU2: Package temperature/speed normal
jul 25 11:55:49 T480s kernel: CPU1: Package temperature/speed normal
jul 25 11:55:49 T480s kernel: CPU3: Package temperature/speed normal
jul 25 11:55:49 T480s kernel: CPU6: Package temperature/speed normal
jul 25 11:55:49 T480s kernel: CPU7: Package temperature/speed normal
jul 25 11:55:49 T480s kernel: CPU4: Package temperature/speed normal
jul 25 11:55:49 T480s kernel: CPU0: Package temperature/speed normal
```
Power Limit Throttling... not sure. 

Please also check this [post at lenovo-throttling-fix](https://github.com/erpalma/lenovo-throttling-fix/issues/29#issuecomment-418295459) as you two might exchange/combine knowledge & power :)

amanusk, 2018-09-26:
Note: Might be possible by looking `/sys/bus/cpu/devices/cpu0/thermal_throttle/core_throttle_count`

erpalma, 2018-11-07:
I've just added a very basic monitoring of throttling causes, you might have a [look](https://github.com/erpalma/lenovo-throttling-fix/commit/9636a5de8327fc2d6874b97da93e6169d9d78588).

amanusk, 2018-11-07:
@erpalma, thanks for the notice!
I have seen the commit yesterday, defiantly interested in adding this feature :)

LinuxOnTheDesktop, 2022-02-18:
Sorry to ask a question here, and especially on a closed issue, but where in s-tui does one see the throttling information? I have seen it before but for the life of me I cannot seem to see it now!

LinuxOnTheDesktop, 2024-10-23:
Might I repeat my question above? I continue to be unable to find the new(ish) wonderful feature!