amanusk/s-tui #87: Temperature sensors without threshold crash progra,

mkesper, 2018-10-04:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel 64
  * OS version: Debian Buster
  * s-tui version: 0.8.2
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):
Crash of whole application:
  Traceback (most recent call last):                                                                         [245/733]
  File "/home/mkesper/src/mkesper/pip/python/bin/s-tui", line 11, in <module>                                      
    load_entry_point('s-tui==0.8.2', 'console_scripts', 's-tui')()                                                 
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/s_tui.py", line 928, in main         
    graph_controller.main()                                                                                        
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/s_tui.py", line 760, in main
    self.loop.run()                                                                                                 
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/main_loop.py", line 286, in run
    self._run()                                         
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/main_loop.py", line 384, in _run     
    self.event_loop.run()                                
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/main_loop.py", line 788, in run     
    self._loop()                                                                                                    
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/main_loop.py", line 825, in _loop   
    self._watch_files[fd]()                                                                                  
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/raw_display.py", line 404, in <lambda
>                                                                                                                  
    event_loop, callback, self.get_available_raw_input())                                                          
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/raw_display.py", line 502, in parse_i
nput                                                                                                               
    callback(processed, processed_codes)                
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)                                                
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/main_loop.py", line 511, in process_i
nput                                                                                     
    k = self._topmost_widget.keypress(self.screen_size, k)             
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)                                                              
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)                                  
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/UiElements.py", line 37, in keypress
    return super(ViListBox, self).keypress(size, key)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/urwid/signals.py", line 295, in _call_callb
ack
    return bool(callback(*args_to_pass))
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/TempSensorsMenu.py", line 114, in on_
apply
    self.return_fn()
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/s_tui.py", line 289, in on_sensors_me
nu_close
    self.__init__(self.controller)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/s_tui.py", line 233, in __init__
    urwid.WidgetPlaceholder.__init__(self, self.main_window())
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/s_tui.py", line 552, in main_window    self.controller.temp_thresh)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/Sources/TemperatureSource.py", line 5
4, in __init__
    self.update()
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/Sources/TemperatureSource.py", line 1
07, in update
    update_func(sensor_major, int(sensor_minor))
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/Sources/TemperatureSource.py", line 9
3, in update_func
    Source.update(self)
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/Sources/Source.py", line 28, in updat
e
    self.eval_hooks()
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/Sources/Source.py", line 68, in eval_
hooks
    if self.get_edge_triggered():
  File "/home/mkesper/src/mkesper/pip/python/lib/python3.6/site-packages/s_tui/Sources/TemperatureSource.py", line 1
90, in get_edge_triggered
    return self.last_temp > self.temp_thresh
TypeError: '>' not supported between instances of 'float' and 'NoneType'

#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 

[_s-tui.log](https://github.com/amanusk/s-tui/files/2448136/_s-tui.log)
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Choose a temp. sensor without threshhold set, e.g. pch_skylake, 0
2. Apply
3. Crash

mkesper, 2018-10-05:
See #88 and #89 respectively for possible approaches.

amanusk, 2018-10-06:
Thanks for opening the issue.
This indeed happens when `psutil` does not return `max` or `critical` temperatures, which is possible.

closed 2018-11-30 by amanusk