amanusk/s-tui #90: No valid Python3 package

mkesper, 2018-10-05:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: Debian Buster
  * s-tui version: 0.8.2
  * Installation method(pip/PPA/source code): Git clone
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):
  
If called as a module with Python2, the program will start, but Python3 does not find it as a module:

```
mkesper@debian:~/src/mkesper/s-tui (temp_threshold_zero) $ python2 --version
Python 2.7.15+
mkesper@debian:~/src/mkesper/s-tui (temp_threshold_zero) $ python3 --version
Python 3.6.6
mkesper@debian:~/src/mkesper/s-tui (temp_threshold_zero) $ python2 -m s_tui/s_tui --version
s-tui 0.8.2 - (C) 2017-2018 Alex Manuskin, Gil Tsuker
    Released under GNU GPLv2
mkesper@debian:~/src/mkesper/s-tui (temp_threshold_zero) $ python3 -m s_tui/s_tui --version
/usr/bin/python3: No module named s_tui/s_tui
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Git clone the source
2. Enter the source directory
3. Execute

```
python3 -m s_tui/s_tui --version
```

amanusk, 2018-10-06:
To my knowledge, the way to run modules with `-m` is by separating them with `.` instead of `
/`
Both `python3 -m s_tui.s_tui --version` and `python2 -m s_tui.s_tui --version` should work. 
Or is there a case where `s_tui/s_tui` is needed?

mkesper, 2018-10-07:
You're right. My bad, python2 seems to be more lenient here.

closed 2018-10-07 by mkesper