amanusk/s-tui #93: stress process keeps running after a crash while in stress mode

christopher-dG, 2018-10-08:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel i7 6700K
  * OS version: Arch Linux
```
$ uname -a
Linux shoebox 4.18.10-arch1-1-ARCH #1 SMP PREEMPT Wed Sep 26 09:48:22 UTC 2018 x86_64 GNU/Linux
```
  * s-tui version: 0.8.2
  * Installation method(pip/PPA/source code): AUR
  
### Step 2: Describe the problem:
When the program crashes in stress mode, `stress` keeps running in the background, keeping the CPU pinned.
#### Observed Results:
I'm using #92 to reproduce this.
```
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 11, in <module>
    load_entry_point('s-tui==0.8.2', 'console_scripts', 's-tui')()
  File "/usr/lib/python3.7/site-packages/s_tui/s_tui.py", line 928, in main
    graph_controller.main()
  File "/usr/lib/python3.7/site-packages/s_tui/s_tui.py", line 760, in main
    self.loop.run()
  File "/usr/lib/python3.7/site-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3.7/site-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3.7/site-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3.7/site-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3.7/site-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3.7/site-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3.7/site-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3.7/site-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3.7/site-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3.7/site-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3.7/site-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3.7/site-packages/s_tui/UiElements.py", line 37, in keypress
    return super(ViListBox, self).keypress(size, key)
  File "/usr/lib/python3.7/site-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3.7/site-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3.7/site-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/usr/lib/python3.7/site-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3.7/site-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3.7/site-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/lib/python3.7/site-packages/s_tui/TempSensorsMenu.py", line 114, in on_apply
    self.return_fn()
  File "/usr/lib/python3.7/site-packages/s_tui/s_tui.py", line 289, in on_sensors_menu_close
    self.__init__(self.controller)
  File "/usr/lib/python3.7/site-packages/s_tui/s_tui.py", line 233, in __init__
    urwid.WidgetPlaceholder.__init__(self, self.main_window())
  File "/usr/lib/python3.7/site-packages/s_tui/s_tui.py", line 552, in main_window
    self.controller.temp_thresh)
  File "/usr/lib/python3.7/site-packages/s_tui/Sources/TemperatureSource.py", line 54, in __init__
    self.update()
  File "/usr/lib/python3.7/site-packages/s_tui/Sources/TemperatureSource.py", line 107, in update
[... 106 lines cut ...]
2018-10-08 12:24:22,353 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:24:22,353 [__init__()] [DEBUG]  
2018-10-08 12:24:22,354 [__init__()] [INFO ]  num cpus 8
2018-10-08 12:24:22,493 [update()] [INFO ]  Utilization recorded 100.0
2018-10-08 12:24:22,493 [__init__()] [DEBUG]  arg temp  pch_skylake,0,
2018-10-08 12:24:22,493 [init_update()] [DEBUG]  custom temp is pch_skylake,0,
2018-10-08 12:24:22,493 [init_update()] [DEBUG]  Selected custom temp
2018-10-08 12:24:22,493 [init_update()] [DEBUG]  Major pch_skylake Minor 0
2018-10-08 12:24:22,495 [set_threshold()] [DEBUG]  Temperature threshold set to None
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. start s-tui
2. enter stress mode
3. find a way to make it crash
4. observe stress still running

```sh
$ s-tui
# Make it crash
$ pidof stress
27898 27897 27896 27895 27894 27893 27892 27891 27890
$ pkill stress
```

amanusk, 2018-10-15:
Thanks for opening the issue.
PR #89  should fix the problem with a threshold value set to None, and in turn there should be not crash with stress processes running. Could you please help test this by pulling from master?

amanusk, 2018-11-30:
Assuming this is closed. Please reopen if needed

closed 2018-11-30 by amanusk