amanusk/s-tui #94: Monitor/stress mode checkboxes are both unchecked after changing temperature sensor

christopher-dG, 2018-10-08:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel i7 6700K
  * OS version: Arch Linux
```
$ uname -a
Linux shoebox 4.18.10-arch1-1-ARCH #1 SMP PREEMPT Wed Sep 26 09:48:22 UTC 2018 x86_64 GNU/Linux
```
  * s-tui version: 0.8.2
  * Installation method(pip/PPA/source code): AUR
  
### Step 2: Describe the problem:
When you change the temperature sensor, the checkboxes that indicate whether we're in monitor or stress mode are both cleared, regardless of which one you were in before.
#### Observed Results:
[image]

#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2018-10-08 12:33:19,545 [main()] [INFO ]  Started without root permissions
2018-10-08 12:33:19,545 [__init__()] [DEBUG]  Config file not found
2018-10-08 12:33:19,545 [__init__()] [DEBUG]  No refresh rate configed
2018-10-08 12:33:19,545 [__init__()] [DEBUG]  No user config for utf8
2018-10-08 12:33:19,545 [__init__()] [DEBUG]  No user config for temp sensor
2018-10-08 12:33:19,545 [__init__()] [DEBUG]  No user config for temp threshold
2018-10-08 12:33:19,547 [__init__()] [DEBUG]  stress-ng is not installed
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Package id 0
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Core 0
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Core 1
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Core 2
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Core 3
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  Sensor Label
2018-10-08 12:33:19,551 [__init__()] [DEBUG]  
2018-10-08 12:33:19,552 [__init__()] [INFO ]  num cpus 8
2018-10-08 12:33:19,696 [update()] [INFO ]  Utilization recorded 9.5
2018-10-08 12:33:19,696 [__init__()] [DEBUG]  arg temp  None
2018-10-08 12:33:19,696 [init_update()] [DEBUG]  custom temp is None
2018-10-08 12:33:19,703 [init_update()] [DEBUG]  Temperature sensor is set to coretemp
2018-10-08 12:33:19,708 [set_threshold()] [DEBUG]  Temperature threshold set to 80.0
2018-10-08 12:33:19,715 [__init__()] [DEBUG]  Update is updated to <function TemperatureSource.init_update.<locals>.update at 0x7f50e74f7ea0>
2018-10-08 12:33:19,716 [get_power_usage()] [INFO ]  current 131436500313.0 last 131436500313.0
2018-10-08 12:33:19,716 [get_power_usage()] [INFO ]  Joule_Used 0.0 seconds_passed 0.0005950927734375
2018-10-08 12:33:19,719 [update()] [DEBUG]  Fan Speend Not Available
2018-10-08 12:33:19,719 [update()] [INFO ]  Fan speed recorded0.0
2018-10-08 12:33:19,752 [on_unicode_checkbox()] [DEBUG]  unicode State is False
2018-10-08 12:33:19,760 [update()] [INFO ]  Utilization recorded 16.3
2018-10-08 12:33:19,766 [get_power_usage()] [INFO ]  current 131436945625.0 last 131436500313.0
2018-10-08 12:33:19,766 [get_power_usage()] [INFO ]  Joule_Used 0.445312 seconds_passed 0.04976630210876465
2018-10-08 12:33:19,766 [get_power_usage()] [INFO ]  Power reading elapsed
2018-10-08 12:33:19,766 [get_power_usage()] [INFO ]  Max power updated 9
2018-10-08 12:33:19,766 [update_displayed_graph_data()] [INFO ]  Reading 800
2018-10-08 12:33:19,767 [update_displayed_graph_data()] [INFO ]  Reading 16.3
2018-10-08 12:33:19,767 [update_displayed_graph_data()] [INFO ]  Reading 36.0
2018-10-08 12:33:19,768 [update_displayed_graph_data()] [INFO ]  Reading 8.948062868460008
2018-10-08 12:33:19,771 [update()] [INFO ]  Utilization recorded 11.1
2018-10-08 12:33:19,775 [get_power_usage()] [INFO ]  current 131437039374.0 last 131436945625.0
2018-10-08 12:33:19,775 [get_power_usage()] [INFO ]  Joule_Used 0.093749 seconds_passed 0.008836030960083008
2018-10-08 12:33:19,775 [get_power_usage()] [INFO ]  Power reading elapsed
2018-10-08 12:33:19,775 [get_power_usage()] [INFO ]  Max power updated 11
2018-10-08 12:33:19,775 [update_displayed_graph_data()] [INFO ]  Reading 1023
[... 47 lines cut ...]
2018-10-08 12:33:26,280 [update()] [INFO ]  Utilization recorded 14.3
2018-10-08 12:33:26,280 [__init__()] [DEBUG]  arg temp  acpitz,1,
2018-10-08 12:33:26,281 [init_update()] [DEBUG]  custom temp is acpitz,1,
2018-10-08 12:33:26,281 [init_update()] [DEBUG]  Selected custom temp
2018-10-08 12:33:26,281 [init_update()] [DEBUG]  Major acpitz Minor 1
2018-10-08 12:33:26,283 [set_threshold()] [DEBUG]  Temperature threshold set to 119.0
2018-10-08 12:33:26,285 [__init__()] [DEBUG]  Update is updated to <function TemperatureSource.init_update.<locals>.update at 0x7f50e72a77b8>
2018-10-08 12:33:26,286 [get_power_usage()] [INFO ]  current 131491748402.0 last 131491748402.0
2018-10-08 12:33:26,286 [get_power_usage()] [INFO ]  Joule_Used 0.0 seconds_passed 0.0003230571746826172
2018-10-08 12:33:26,287 [update()] [DEBUG]  Fan Speend Not Available
2018-10-08 12:33:26,287 [update()] [INFO ]  Fan speed recorded0.0
2018-10-08 12:33:26,291 [on_unicode_checkbox()] [DEBUG]  unicode State is False
2018-10-08 12:33:26,293 [on_sensors_menu_close()] [INFO ]  Temp sensor updated to acpitz,1,
2018-10-08 12:33:27,790 [unhandled_input()] [DEBUG]  Caught q
2018-10-08 12:33:27,790 [unhandled_input()] [DEBUG]  None
2018-10-08 12:33:27,790 [kill_child_processes()] [DEBUG]  Killing stress process
2018-10-08 12:33:27,790 [kill_child_processes()] [DEBUG]  No such process
2018-10-08 12:33:27,790 [kill_child_processes()] [DEBUG]  Could not kill process
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. start s-tui
2. change temperature sensor
3. observe unfilled checkboxes

amanusk, 2019-02-05:
Thanks @christopher-dG.
The sensors menu is planned to be overhauled in the next version, which should also address this.
(You can check the progress on the develop branch, it is far from ready)

amanusk, 2019-04-01:
@christopher-dG
The master was update to a new version, where the issue should be fixed.

Help testing is appreciated :smiley:

amanusk, 2020-01-23:
This should now be fixed in v1.0

closed 2020-01-23 by amanusk