amanusk/s-tui #95: Temperature: wrong value is displayed or - more likely - the temperature is displayed in a way apt to be misread

CottonEaster, 2018-10-14:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): 20KHCTO1WW ThinkPad X1 Carbon 6th, Intel i7-8550U (8) @ 4.000GHz
  * OS version: Mint 19 x64 Cinnamon
  * s-tui version:  0.8.2
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results: My conky and the program `hardinfo` show the cpu temperature as being significantly less (namely, some 70 degrees) than s-tui seems to report (namely, some 85-90 degrees). _However_, I think I may be reading s-tui's graph wrongly.

On all this, please see attached screenshot, and note that the 'spikes' on s-tui's temperature chart surprised me, in that I thought they were off the scale. Perhaps some sort of graphical notch, showing the ends of the scale, would be an idea? Also, showing the current temperature, as a number, would be good - but perhaps there is a way of doing that already.

  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Run s-tui stress test (though perhaps my specific hardware plays a role?).
2. Compare what s-tui's temperature graph seems to say, to what other temperature monitoring tools say.

Screenshot:
[image]

CottonEaster, 2018-10-14:
I did have s-tui in a somewhat larger window; the window is very small in the screenshot because I wanted to get other windows into the shot. But I take your point - about needing a large-ish window to see a decent amount of detail, about the scrolling (about which, yes, I was unaware). Still, I continue to think that notches on the scale might be a good idea.

CottonEaster, 2018-10-14:
Also, is the 'FAQ' link broken? Nothing seems to happen when I click it (and I tried in two browsers).

amanusk, 2018-10-14:
We did try to add additional notches, when the screen is enlarged.

[image]

> Also, is the 'FAQ' link broken? Nothing seems to happen when I click it (and I tried in two browsers). 

Yes, it appears so

CottonEaster, 2018-10-14:
Re notches: I do not mean values on an axis. I mean short horizontal lines connecting values on an axis with the body of the graph - somewhat like this:

_
|
|
|

See also [this graph](http://cdn.pythagorasandthat.co.uk/wp-content/uploads/2014/07/quadratic-graph.png).