amanusk/s-tui #97: Request function command to disable uti,cur,max from terminal

ryanmusante, 2018-10-21:
### Step 1: Describe your environment
```
ryan@xmg Linux 4.18.16-31-tkg-pds x86_64 18.0.0-rc Illyria
s-tui 0.8.2 - (C) 2017-2018 Alex Manuskin, Gil Tsuker
Installation method: yay -S
```
### Step 2: Describe the problem:
Request to disable utilization, cur power and max power from "-t" readings on termianl
```
Utilization: 0.0 %, Cur Power: 13.9 W, Max Power: 14.0 W,
```
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`:

amanusk, 2018-10-21:
Hi, could you please elaborate more on what the feature request is?

ryanmusante, 2018-10-23:
Is it possible to have the "-t" terminal command show less information? 

I'm only interested in seeing a few parts because util/curpower/maxpower seem to use a lot of cpu resources constantly polling.

amanusk, 2018-10-23:
I see, there is no way at the moment, but I would love to know how you determined which calls use most CPU, we would like to improve performance in upcoming versions.

closed 2018-12-28 by ryanmusante