amanusk/s-tui #100: RuntimeWarning; OSError 61

CottonEaster, 2018-11-19:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Inteli7-8550U, on ThinkPad X1CG6
  * OS version: Mint 19 Cinnamon
  * s-tui version: 0.8.2
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):

```
 #  s-tui
/usr/lib/python3/dist-packages/psutil/_pslinux.py:1150: RuntimeWarning: ignoring OSError(61, 'No data available')
  warnings.warn("ignoring %r" % err, RuntimeWarning)
```
though the program ran. Another error appeared when I ran the stress test, but the test did run.
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2018-11-19 21:41:34,865 [main()] [INFO ]  Started without root permissions
2018-11-19 21:41:34,866 [__init__()] [DEBUG]  Config file not found
2018-11-19 21:41:34,866 [__init__()] [DEBUG]  No refresh rate configed
2018-11-19 21:41:34,866 [__init__()] [DEBUG]  No user config for utf8
2018-11-19 21:41:34,866 [__init__()] [DEBUG]  No user config for temp sensor
2018-11-19 21:41:34,866 [__init__()] [DEBUG]  No user config for temp threshold
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Sensor Label
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Sensor Label
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Package id 0
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Sensor Label
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Core 0
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Sensor Label
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Core 1
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Sensor Label
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Core 2
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Sensor Label
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Core 3
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  Sensor Label
2018-11-19 21:41:34,873 [__init__()] [DEBUG]  
2018-11-19 21:41:34,874 [__init__()] [INFO ]  num cpus 8
2018-11-19 21:41:35,014 [update()] [INFO ]  Utilization recorded 21.8
2018-11-19 21:41:35,014 [__init__()] [DEBUG]  arg temp  None
2018-11-19 21:41:35,014 [init_update()] [DEBUG]  custom temp is None
2018-11-19 21:41:35,017 [init_update()] [DEBUG]  Temperature sensor is set to coretemp
2018-11-19 21:41:35,020 [set_threshold()] [DEBUG]  Temperature threshold set to 100.0
2018-11-19 21:41:35,023 [__init__()] [DEBUG]  Update is updated to <function TemperatureSource.init_update.<locals>.update at 0x7f1aa2e1a620>
2018-11-19 21:41:35,023 [get_power_usage()] [INFO ]  current 55543169881.0 last 55543169881.0
2018-11-19 21:41:35,023 [get_power_usage()] [INFO ]  Joule_Used 0.0 seconds_passed 0.00011563301086425781
2018-11-19 21:41:35,025 [update()] [INFO ]  Fan speed recorded0.0
2018-11-19 21:41:35,030 [on_unicode_checkbox()] [DEBUG]  unicode State is False
2018-11-19 21:41:35,034 [update()] [INFO ]  Utilization recorded 6.7
2018-11-19 21:41:35,037 [get_power_usage()] [INFO ]  current 55543232869.0 last 55543169881.0
2018-11-19 21:41:35,037 [get_power_usage()] [INFO ]  Joule_Used 0.062988 seconds_passed 0.013588905334472656
2018-11-19 21:41:35,037 [get_power_usage()] [INFO ]  Power reading elapsed
2018-11-19 21:41:35,037 [get_power_usage()] [INFO ]  Max power updated 5
2018-11-19 21:41:35,038 [update()] [INFO ]  Fan speed recorded0.0
2018-11-19 21:41:35,038 [update_displayed_graph_data()] [INFO ]  Reading 2400
2018-11-19 21:41:35,038 [update_displayed_graph_data()] [INFO ]  Reading 6.7
2018-11-19 21:41:35,038 [update_displayed_graph_data()] [INFO ]  Reading 61.0
2018-11-19 21:41:35,039 [update_displayed_graph_data()] [INFO ]  Reading 4.63525195368096
2018-11-19 21:41:35,041 [update()] [INFO ]  Utilization recorded 14.3
2018-11-19 21:41:35,044 [get_power_usage()] [INFO ]  current 55543274007.0 last 55543232869.0
2018-11-19 21:41:35,044 [get_power_usage()] [INFO ]  Joule_Used 0.041138 seconds_passed 0.007486581802368164
2018-11-19 21:41:35,044 [get_power_usage()] [INFO ]  Power reading elapsed
2018-11-19 21:41:35,044 [get_power_usage()] [INFO ]  Max power updated 6
2018-11-19 21:41:35,045 [update()] [INFO ]  Fan speed recorded0.0
2018-11-19 21:41:35,045 [update_displayed_graph_data()] [INFO ]  Reading 2400
2018-11-19 21:41:35,046 [update_displayed_graph_data()] [INFO ]  Reading 14.3
2018-11-19 21:41:35,046 [update_displayed_graph_data()] [INFO ]  Reading 61.0
2018-11-19 21:41:35,046 [update_displayed_graph_data()] [INFO ]  Reading 5.494897549504793
2018-11-19 21:41:37,189 [update()] [INFO ]  Utilization recorded 8.4
2018-11-19 21:41:37,198 [get_power_usage()] [INFO ]  current 55550102908.0 last 55543274007.0
2018-11-19 21:41:37,198 [get_power_usage()] [INFO ]  Joule_Used 6.828901 seconds_passed 2.153468370437622
2018-11-19 21:41:37,198 [get_power_usage()] [INFO ]  Power reading elapsed
2018-11-19 21:41:37,201 [update()] [INFO ]  Fan speed recorded0.0
2018-11-19 21:41:37,201 [update_displayed_graph_data()] [INFO ]  Reading 2271
2018-11-19 21:41:37,202 [update_displayed_graph_data()] [INFO ]  Reading 8.4
2018-11-19 21:41:37,203 [update_displayed_graph_data()] [INFO ]  Reading 45.0
2018-11-19 21:41:37,204 [update_displayed_graph_data()] [INFO ]  Reading 3.1711173907849175
2018-11-19 21:41:39,353 [update()] [INFO ]  Utilization recorded 12.8
2018-11-19 21:41:39,362 [get_power_usage()] [INFO ]  current 55558129011.0 last 55550102908.0
2018-11-19 21:41:39,362 [get_power_usage()] [INFO ]  Joule_Used 8.026103 seconds_passed 2.164008140563965
2018-11-19 21:41:39,362 [get_power_usage()] [INFO ]  Power reading elapsed
2018-11-19 21:41:39,365 [update()] [INFO ]  Fan speed recorded0.0
2018-11-19 21:41:39,365 [update_displayed_graph_data()] [INFO ]  Reading 829
2018-11-19 21:41:39,366 [update_displayed_graph_data()] [INFO ]  Reading 12.8
2018-11-19 21:41:39,367 [update_displayed_graph_data()] [INFO ]  Reading 45.0
2018-11-19 21:41:39,368 [update_displayed_graph_data()] [INFO ]  Reading 3.7089061032405857
2018-11-19 21:41:41,533 [update()] [INFO ]  Utilization recorded 6.0
2018-11-19 21:41:41,541 [get_power_usage()] [INFO ]  current 55561804660.0 last 55558129011.0
2018-11-19 21:41:41,541 [get_power_usage()] [INFO ]  Joule_Used 3.675649 seconds_passed 2.1791880130767822
2018-11-19 21:41:41,541 [get_power_usage()] [INFO ]  Power reading elapsed
2018-11-19 21:41:41,544 [update()] [INFO ]  Fan speed recorded0.0
2018-11-19 21:41:41,544 [update_displayed_graph_data()] [INFO ]  Reading 1403
2018-11-19 21:41:41,545 [update_displayed_graph_data()] [INFO ]  Reading 6.0
2018-11-19 21:41:41,546 [update_displayed_graph_data()] [INFO ]  Reading 45.0
2018-11-19 21:41:41,546 [update_displayed_graph_data()] [INFO ]  Reading 1.6867057720322045
2018-11-19 21:41:43,689 [update()] [INFO ]  Utilization recorded 12.3
2018-11-19 21:41:43,699 [get_power_usage()] [INFO ]  current 55568434709.0 last 55561804660.0
2018-11-19 21:41:43,699 [get_power_usage()] [INFO ]  Joule_Used 6.630049 seconds_passed 2.1580300331115723
2018-11-19 21:41:43,700 [get_power_usage()] [INFO ]  Power reading elapsed
2018-11-19 21:41:43,702 [update()] [INFO ]  Fan speed recorded0.0
2018-11-19 21:41:43,702 [update_displayed_graph_data()] [INFO ]  Reading 885
2018-11-19 21:41:43,703 [update_displayed_graph_data()] [INFO ]  Reading 12.3
2018-11-19 21:41:43,705 [update_displayed_graph_data()] [INFO ]  Reading 45.0
2018-11-19 21:41:43,705 [update_displayed_graph_data()] [INFO ]  Reading 3.0722691057456752
2018-11-19 21:41:44,293 [unhandled_input()] [DEBUG]  Caught q
2018-11-19 21:41:44,293 [unhandled_input()] [DEBUG]  None
2018-11-19 21:41:44,293 [kill_child_processes()] [DEBUG]  Killing stress process
2018-11-19 21:41:44,294 [kill_child_processes()] [DEBUG]  No such process
2018-11-19 21:41:44,294 [kill_child_processes()] [DEBUG]  Could not kill process

```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Run s-tui, as root or not, from terminal.

But - ah - same as #29 ?

amanusk, 2018-11-20:
Hi @CottonEaster, so the program was running and the error showed up after you have closed it?
Did any of the graphs show incorrect stats?
Also, what version of `psutil` are you using?
Check with
` python -c "import psutil; print(psutil.__version__)"`

amanusk, 2019-04-01:
Closing for not response

closed 2019-04-01 by amanusk