amanusk/s-tui #103: // , pip install --user s-tui fails on a fresh ubuntu installation, just installed pip

v6, 2018-12-13:
// , 
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: 18.10
  * s-tui version: Latest
  * Installation method(pip/PPA/source code): `pip install --user s-tui`
  
### Step 2: Describe the problem:

Installation requires `sudo`

#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):

```
nb@nb-ThinkPad-X1-Carbon-6th:~$ pip install s-tui
Collecting s-tui
  Downloading https://files.pythonhosted.org/packages/f4/43/d02518a7a3c178b04adbc813ed0959c7d89bff0fd6921286bb431de5649a/s-tui-0.8.3.tar.gz
Collecting urwid>=1.3.1 (from s-tui)
  Downloading https://files.pythonhosted.org/packages/c7/90/415728875c230fafd13d118512bde3184d810d7bf798a631abc05fac09d0/urwid-2.0.1.tar.gz (604kB)
    100% |████████████████████████████████| 614kB 172kB/s 
Collecting psutil>=5.2.0 (from s-tui)
  Downloading https://files.pythonhosted.org/packages/e3/58/0eae6e4466e5abf779d7e2b71fac7fba5f59e00ea36ddb3ed690419ccb0f/psutil-5.4.8.tar.gz (422kB)
    100% |████████████████████████████████| 430kB 10.1MB/s 
Building wheels for collected packages: s-tui, urwid, psutil
  Running setup.py bdist_wheel for s-tui ... done
  Stored in directory: /home/nb/.cache/pip/wheels/3e/25/98/b0999024f1cba804214627f38d1dccaaa7f6b6e0dfefa429e4
  Running setup.py bdist_wheel for urwid ... done
  Stored in directory: /home/nb/.cache/pip/wheels/34/06/50/24011714e101b5ad8518c69175d117a257413e3c15a92983f0
  Running setup.py bdist_wheel for psutil ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-fTerZq/psutil/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-jDwtfy --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-2.7
  creating build/lib.linux-x86_64-2.7/psutil
  copying psutil/_common.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_psosx.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_pswindows.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_compat.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_pssunos.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_psposix.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_pslinux.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_psbsd.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/__init__.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_psaix.py -> build/lib.linux-x86_64-2.7/psutil
  copying psutil/_exceptions.py -> build/lib.linux-x86_64-2.7/psutil
  creating build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_bsd.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_process.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_system.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_contracts.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_aix.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_linux.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_misc.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/__init__.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_osx.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_posix.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_sunos.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_memory_leaks.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/__main__.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_unicode.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_connections.py -> build/lib.linux-x86_64-2.7/psutil/tests
  copying psutil/tests/test_windows.py -> build/lib.linux-x86_64-2.7/psutil/tests
  running build_ext
  building 'psutil._psutil_linux' extension
  creating build/temp.linux-x86_64-2.7
[... 158 lines cut ...]
  --client-cert <path>        Path to SSL client certificate, a single file
                              containing the private key and the certificate
                              in PEM format.
  --cache-dir <dir>           Store the cache data in <dir>.
  --no-cache-dir              Disable the cache.
  --disable-pip-version-check
                              Don't periodically check PyPI to determine
                              whether a new version of pip is available for
                              download. Implied with --no-index.
  --no-color                  Suppress colored output
nb@nb-ThinkPad-X1-Carbon-6th:~$ 
```
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
Can't run these, since I can't install. 
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Attempt to install `s-tui` as a non-root user.

```
pip install --user s-tui
```

amanusk, 2018-12-15:
Thanks for the issue, 
It seems the problem is with the installation of `psutil`
Could you please try to install only `psutil` with pip and see if the issue persists?
Also, you should be able to install s-tui on `18.04` with simply `sudo apt install s-tui`

amanusk, 2019-01-19:
@v6  sorry for the mix-up, installation with apt without the PPA is only available from `18.10` and onward.   
Also, you should checkout https://github.com/erpalma/throttled for mitigating power management issues.

nayr1991, 2019-03-25:
I had this same issue on centos7 and it was due to missing python-devel headers.
Resolving it by installing python-devel with yum

iunlock, 2019-07-01:
> Thanks for the issue,
> It seems the problem is with the installation of `psutil`
> Could you please try to install only `psutil` with pip and see if the issue persists?
> Also, you should be able to install s-tui on `18.04` with simply `sudo apt install s-tui`

Hello, I too have a similar issue as the OP. I've tried as you've suggested;

`$ pip install psutil
Collecting psutil
Installing collected packages: psutil
Successfully installed psutil-5.6.3`

Next I did;

`$ sudo apt install s-tui
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package s-tui`

Shouldn't the s-tui package be there? What would cause it to not be there?

I've also tried;

`$ pip install lxml
Collecting lxml
  Downloading https://files.pythonhosted.org/packages/af/31/47cce58942bbf4b8f1c975ec2d1ab52141f7b7cf8cdecb58f25546d2c4fd/lxml-4.3.4-cp27-cp27mu-manylinux1_x86_64.whl (5.6MB)
    100% |████████████████████████████████| 5.6MB 227kB/s 
Installing collected packages: lxml
Successfully installed lxml-4.3.4`

... in order to update the headers... 

I'm on Linux MINT 19.1. Thanks for this app...hope to get it working soon. :)

amanusk, 2019-07-01:
Hi @iunlock,
Unfortunately it seems s-tu is not yet available from mint repositories.
You can install it with `pip`, just like you installed psutil.
If you want to try out the beta version, install with
`pip install s-tui==1.0.0b1`

Please report if this works out for you

iunlock, 2019-07-01:
> Hi @iunlock,
> Unfortunately it seems s-tu is not yet available from mint repositories.
> You can install it with `pip`, just like you installed psutil.
> If you want to try out the beta version, install with
> `pip install s-tui==1.0.0b1`
> 
> Please report if this works out for you

Thanks for the quite reply Dev. :)

`$ pip install s-tui==1.0.0b1
Collecting s-tui==1.0.0b1
  Downloading https://files.pythonhosted.org/packages/6d/a0/fb3676a2cb39b85722d85976cdd7a10881fff9a75978a97448662c42bbe6/s-tui-1.0.0b1.tar.gz
Collecting urwid>=2.0.1 (from s-tui==1.0.0b1)
Collecting psutil>=5.6.0 (from s-tui==1.0.0b1)
Building wheels for collected packages: s-tui
  Running setup.py bdist_wheel for s-tui ... done
  Stored in directory: /home/lg/.cache/pip/wheels/b6/83/43/5b56bb0373c85a35bd3b56f9071519f732b4b90c5c7ff2cb4f
Successfully built s-tui
Installing collected packages: urwid, psutil, s-tui
Successfully installed psutil-5.6.3 s-tui-1.0.0b1 urwid-2.0.1`

...

`$ s-tui
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 9, in <module>
    load_entry_point('s-tui==0.7.6', 'console_scripts', 's-tui')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 43, in <module>
    from distutils.spawn import find_executable
ModuleNotFoundError: No module named 'distutils.spawn'
`
I assume now I would have to install the distutils.spawn?

amanusk, 2019-07-01:
It seems you already have an older version of s-tui installed with `apt` (0.7.6)
You should probably uninstall it first to avoid conflicts (`sudo apt remove --purge s-tui`).

pip executables are usually installed in `~/.local/bin/`

distutils.spawn should not be a dependency in the new version

iunlock, 2019-07-01:
When running that command, this is what it outputs:

`$ sudo apt remove --purge s-tui
      
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package s-tui`

amanusk, 2019-07-01:
Hmm.. this is indeed a bit frustrating. 
The reason is basically that there are several ways to install python packages. Ubuntu/mint has there own packages for psutil for example (python-psutil), but these are often outdated, so pip is an option as well.

Here are the steps I would try:
`sudo apt remove --purge s-tui`
`sudo apt remove --purge python-s-tui`
`sudo apt remove --purge python3-s-tui`
`sudo pip uninstall s-tui`
`sudo pip3 uninstall s-tui`
`pip uninstall s-tui`
`pip3 uninstall s-tui`

Then install with pip

iunlock, 2019-07-02:
Thanks. Here are the results in order and what it's outputting:

`$ sudo apt remove --purge s-tui
[sudo] password for lg:          
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package s-tui`

`$ sudo apt remove --purge python-s-tui
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python-s-tui`

`$ sudo apt remove --purge python3-s-tui
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  python3-urwid
Use 'sudo apt autoremove' to remove it.
The following packages will be REMOVED:
  python3-s-tui*
0 upgraded, 0 newly installed, 1 to remove and 3 not upgraded.
After this operation, 155 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 320609 files and directories currently installed.)
Removing python3-s-tui (0.7.6-1) ...`

`$ sudo pip uninstall s-tui
The directory '/home/xx/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Uninstalling s-tui-1.0.0b1:
  /home/xx/.local/bin/s-tui
  /home/xx/.local/lib/python2.7/site-packages/s_tui-1.0.0b1.dist-info/INSTALLER
  /home/xx/.local/lib/python2.7/site-packages/s_tui-1.0.0b1.dist-info/METADATA
  /home/xx/.local/lib/python2.7/site-packages/s_tui-1.0.0b1.dist-info/RECORD
  /home/xx/.local/lib/python2.7/site-packages/s_tui-1.0.0b1.dist-info/WHEEL
  /home/xx/.local/lib/python2.7/site-packages/s_tui-1.0.0b1.dist-info/entry_points.txt
  /home/xx/.local/lib/python2.7/site-packages/s_tui-1.0.0b1.dist-info/top_level.txt
  /home/xx/.local/lib/python2.7/site-packages/s_tui/__init__.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/__init__.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/about_menu.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/about_menu.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/help_menu.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/help_menu.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/helper_functions.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/helper_functions.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/s_tui.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/s_tui.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sensors_menu.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sensors_menu.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/__init__.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/__init__.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/fan_source.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/fan_source.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/freq_source.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/freq_source.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/hook.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/hook.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/hook_script.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/hook_script.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/rapl_power_source.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/rapl_power_source.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/rapl_read.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/rapl_read.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/script_hook_loader.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/script_hook_loader.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/source.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/sources/source.pyc
[... 52 lines cut ...]
  /home/xx/.local/lib/python2.7/site-packages/s_tui/Sources/UtilSource.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/Sources/__init__.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/Sources/__init__.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/StressMenu.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/StressMenu.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/StuiBarGraph.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/StuiBarGraph.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/SummaryTextList.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/SummaryTextList.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/TempSensorsMenu.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/TempSensorsMenu.pyc
  /home/xx/.local/lib/python2.7/site-packages/s_tui/UiElements.py
  /home/xx/.local/lib/python2.7/site-packages/s_tui/UiElements.pyc
Proceed (y/n)? y
  Successfully uninstalled s-tui-0.8.3`

`$ pip3 uninstall s-tui
Command pip3 not found, but can be installed with:
sudo apt install python3-pip`

I hope the data above helps narrow down the issue. 

I've refrained from installing with pip due to some of the outputs above and wanted to wait to hear back from you before I proceeded. 

Thanks for all your help with this.

amanusk, 2019-07-02:
I didn't know which package manager you used for the old installation, so I listed them all.

It was successfully uninstalled in this stage :)
> sudo apt remove --purge python3-s-tui Reading package lists... Done Building dependency tree Reading state information... Done The following package was automatically installed and is no longer [...]

Package management with python is a bit of a mess, and the whole python2/3 mess does not help.

Some solutions make dependencies easier to deal with by using virtual environments per package, for example [pipsi](https://github.com/mitsuhiko/pipsi) and its successor. These are suitable for installing executables such as `s-tui` and are usually recommended.

You should use whatever method is most comfortable for you.

To install s-tui with pip run:
`pip install s-tui==1.0.0b1`
Your distro probably has some version of pip built in (find out with `which pip`)

amanusk, 2019-07-05:
@iunlock, did it work for you?

closed 2020-07-13 by amanusk