amanusk/s-tui #104: Wrong power being read on Intel system (lower then reported by powertop)

avdudchenko, 2019-01-01:
Hi, I observe that this app reads substantially lower power use then shown by powertop, or power statistic monitor. (for example powertop reports 7.33 W use while s-tui reports 2.2 W). Not sure if there is away to address this of this is expected behavior. 
This is on Ubuntu 18.04 running on lenovo x1 carbon gen 6th.

amanusk, 2019-01-01:
Hi @avdudchenko, thanks for your interest in the project.
This is expected behaviour. `powertop` monitors the battery of the system, and estimates power usage according to "how much battery power was used in a given time?"
As a result, `powertop` will account for additional components that use power, such as the screen, RAM, SSD etc.
On the other hand `s-tui` reads registers, present on `Intel` CPUs, which give a much more accurate report of the power used specifically by the CPU.
The CPU and the screen are usually the most power consuming components of a laptop, so if you stress your CPU, you will see an increased power usage in both `s-tui` and `powertop`.

avdudchenko, 2019-01-01:
amanusk, thanks. I figured that was the case but was not 100% sure on the behavior. Would it be possible to add an option to read total power use (from powertop for example). I'm sure most laptop users would like to see their total power use live in one place. (I might dig through the code and see if I can do it, but my experience with this type of the systems is limited)

amanusk, 2019-02-05:
Thanks @avdudchenko! (I didn't notice you edit)
Do you have a branch where I can test your working code?

amanusk, 2020-07-13:
Closing this, `s-tui` reading is supposed to be different than power top. 
You are welcome to open a PR to add power top in addition.

closed 2020-07-16 by amanusk