amanusk/s-tui #105: power reading not working

doodie, 2019-01-10:
everything is working great except power. i believe the power requirements were Sandy Bridge and newer. The proc I have is a Sky Lake. 

CentOS Linux release 7.6.1810 (Core)
4.16.8-1.el7.elrepo.x86_64

Intel(R) Xeon(R) CPU E3-1225 v5 @ 3.30GHz

Any ideas on why power option would be "n/a"?

amanusk, 2019-01-10:
Thanks for issue, 
Power usage is calculated by reading information from the directory:
`/sys/class/powercap/intel-rapl/`
Do you have access to it?

amanusk, 2019-01-14:
> is there something that i need to install?

To my knowledge, no. This directory and the information in it should be provided by the kernel.  
I know this is not available on Virtual Machines, but since you are getting temperature, I believe this is not your case.  
Another thing might be worth trying is running `sudo modprobe msr`. This should enable the kernel module which accesses the registers on the CPU. (Usually this is not necessary, and I am not sure this will help).

I believe the problem in CentOS, but I am not familiar enough with it.

If you do find a solution, please post it here.

doodie, 2019-01-14:
after reading [this](https://01.org/rapl-power-meter), found 1 of the 2 parameters in the config and it needs to be enabled. need to do more reading on this first.