amanusk/s-tui #114: Hide sensors in left column

GM-Script-Writer-62850, 2019-04-01:
The current sensor toggle only hides it in the graphical chart, there should also be a option to hide them in the left column, some sensor chips do not have all inputs on the chip used as such some are useless and it would be nice to hide those, i am making this a feature request and not a bug as you may want to have them in the column but not taking space in the chart
a UI like this would work i think, just needs labels indicate for chart and column
[image]

amanusk, 2019-04-09:
Thanks for the suggestion, there indeed should be an option to remove sensors from the sidebar as it can get quite cluttered. 

Would you like to give this a try?

GM-Script-Writer-62850, 2019-04-09:
did i try it right?
```
chad@Z97Killer:~$ cd Downloads/s-tui/
chad@Z97Killer:~/Downloads/s-tui$ git pull
remote: Enumerating objects: 151, done.
remote: Counting objects: 100% (151/151), done.
remote: Compressing objects: 100% (91/91), done.
remote: Total 151 (delta 110), reused 99 (delta 59), pack-reused 0
Receiving objects: 100% (151/151), 56.12 KiB | 1.34 MiB/s, done.
Resolving deltas: 100% (110/110), completed with 11 local objects.
From https://github.com/amanusk/s-tui
   727b7d8..d215046  master          -> origin/master
 * [new branch]      pylint_refactor -> origin/pylint_refactor
Updating 727b7d8..d215046
Fast-forward
 .travis.yml                                        |   2 +-
 README.md                                          |   2 +-
 s_tui/GlobalData.py                                |  36 -
 s_tui/UiElements.py                                |  54 --
 s_tui/{AboutMenu.py => about_menu.py}              |  11 +-
 s_tui/{HelpMenu.py => help_menu.py}                |   7 +-
 s_tui/{HelperFunctions.py => helper_functions.py}  | 167 +++--
 .../{temperaturesource.sh => tempsource.sh}        |   0
 s_tui/s_tui.py                                     | 742 ++++++++++-----------
 s_tui/{SensorsMenu.py => sensors_menu.py}          |   6 +-
 s_tui/{Sources => sources}/__init__.py             |   0
 .../FanSource.py => sources/fan_source.py}         |  44 +-
 .../FreqSource.py => sources/freq_source.py}       |  31 +-
 s_tui/{Sources/Hook.py => sources/hook.py}         |  22 +-
 .../ScriptHook.py => sources/hook_script.py}       |  20 +-
 .../rapl_power_source.py}                          |  18 +-
 s_tui/{Sources => sources}/rapl_read.py            |  60 +-
 .../script_hook_loader.py}                         |   7 +-
 s_tui/{Sources/Source.py => sources/source.py}     |  50 +-
 .../temp_source.py}                                |  32 +-
 .../UtilSource.py => sources/util_source.py}       |  36 +-
 s_tui/{StressMenu.py => stress_menu.py}            |  66 +-
 s_tui/{Tests => sturwid}/__init__.py               |   0
 .../bar_graph_vector.py}                           |  78 +--
 .../complex_bar_graph.py}                          | 113 +---
 .../summary_text_list.py}                          |  20 +-
 s_tui/sturwid/ui_elements.py                       | 105 +++
 s_tui/tests/__init__.py                            |   0
 s_tui/{Tests => tests}/test_util_source.py         |   6 +-
 setup.py                                           |   4 +-
 30 files changed, 772 insertions(+), 967 deletions(-)
 delete mode 100644 s_tui/GlobalData.py
 delete mode 100644 s_tui/UiElements.py
 rename s_tui/{AboutMenu.py => about_menu.py} (90%)
 rename s_tui/{HelpMenu.py => help_menu.py} (89%)
 rename s_tui/{HelperFunctions.py => helper_functions.py} (51%)
 rename s_tui/hooks.d/{temperaturesource.sh => tempsource.sh} (100%)
 rename s_tui/{SensorsMenu.py => sensors_menu.py} (96%)
 rename s_tui/{Sources => sources}/__init__.py (100%)
 rename s_tui/{Sources/FanSource.py => sources/fan_source.py} (66%)
 rename s_tui/{Sources/FreqSource.py => sources/freq_source.py} (70%)
 rename s_tui/{Sources/Hook.py => sources/hook.py} (71%)
 rename s_tui/{Sources/ScriptHook.py => sources/hook_script.py} (75%)
 rename s_tui/{Sources/RaplPowerSource.py => sources/rapl_power_source.py} (88%)
 rename s_tui/{Sources => sources}/rapl_read.py (54%)
 rename s_tui/{Sources/ScriptHookLoader.py => sources/script_hook_loader.py} (91%)
 rename s_tui/{Sources/Source.py => sources/source.py} (70%)
 rename s_tui/{Sources/TemperatureSource.py => sources/temp_source.py} (79%)
 rename s_tui/{Sources/UtilSource.py => sources/util_source.py} (60%)
 rename s_tui/{StressMenu.py => stress_menu.py} (79%)
 rename s_tui/{Tests => sturwid}/__init__.py (100%)
 rename s_tui/{StuiBarGraphVector.py => sturwid/bar_graph_vector.py} (82%)
 rename s_tui/{ComplexBarGraphs.py => sturwid/complex_bar_graph.py} (63%)
 rename s_tui/{SummaryTextList.py => sturwid/summary_text_list.py} (59%)
 create mode 100644 s_tui/sturwid/ui_elements.py
 create mode 100644 s_tui/tests/__init__.py
 rename s_tui/{Tests => tests}/test_util_source.py (68%)
chad@Z97Killer:~/Downloads/s-tui$ python -m s_tui.s_tui
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 865, in <module>
    main()
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 820, in main
    graph_controller.main()
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 682, in main
    self.animate_graph(loop)
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 767, in animate_graph
    self.view.update_displayed_information()
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 237, in update_displayed_information
    graph.update()
  File "s_tui/sturwid/bar_graph_vector.py", line 163, in update
    if self.visible_graph_list[graph_idx]:
IndexError: list index out of range
```

amanusk, 2019-04-09:
Crash is probably due to inconsistencies in config file. Try removing `~/.config/s-tui/s-tui.conf` (this is a separate bug)

> did i try it right?

It seems from your image that you have partially implemented the feature, I was wandering if you would like to open a Pull Request

GM-Script-Writer-62850, 2019-04-09:
So I take it my GIMP skill are impressive then?

Removing said file make it load

When you said give this a try i assumed you had committed a change to implement said feature

amanusk, 2019-04-09:
> So I take it my GIMP skill are impressive then?

Absolutely 

> When you said give this a try i assumed you had committed a change to implement said feature

Yp .. not yet

amanusk, 2019-04-22:
New submenu with option to remove summaries in side bar was added to master

GM-Script-Writer-62850, 2019-04-22:
It works, but where did my horizontal lines go and why are my vertical lines now pipes?
[image]

GM-Script-Writer-62850, 2019-04-22:
save settings is broken that or more likely load settings is
```
cat ~/.config/s-tui/s-tui.conf
[GraphControll]
refresh = 1.0
utf8 = True

[Util,Graphs]
avg = True
core 0 = True
core 1 = True
core 2 = True
core 3 = True

[Frequency,Graphs]
avg = False
core 0 = True
core 1 = True
core 2 = True
core 3 = True

[Fan,Graphs]
nct6791,0 = True
nct6791,1 = True
nct6791,2 = True
nct6791,3 = True
nct6791,4 = False
nct6791,5 = True

[Power,Graphs]
package-0 = True
core,pkg0 = True
dram,pkg0 = True

[Temp,Graphs]
packageid0 = True
core0,pkg0 = True
core1,pkg0 = True
core2,pkg0 = True
core3,pkg0 = True
systin,pkg0 = True
pch_cpu_temp,pkg0 = False
cputin,pkg0 = True
auxtin0,pkg0 = False
auxtin1,pkg0 = False
auxtin2,pkg0 = True
auxtin3,pkg0 = False
peciagent0,pkg0 = True
pch_chip_cpu_max_temp,pkg0 = False
pch_chip_temp,pkg0 = False

[Util,Summaries]
avg = True
core 0 = True
core 1 = True
core 2 = True
core 3 = True

[Frequency,Summaries]
avg = False
core 0 = True
core 1 = True
core 2 = True
core 3 = True

[Fan,Summaries]
nct6791,0 = True
nct6791,1 = True
nct6791,2 = True
nct6791,3 = True
nct6791,4 = False
nct6791,5 = True

[Power,Summaries]
package-0 = True
core,pkg0 = True
dram,pkg0 = True

[Temp,Summaries]
packageid0 = True
core0,pkg0 = True
core1,pkg0 = True
core2,pkg0 = True
core3,pkg0 = True
systin,pkg0 = True
pch_cpu_temp,pkg0 = False
cputin,pkg0 = True
auxtin0,pkg0 = False
auxtin1,pkg0 = False
auxtin2,pkg0 = True
auxtin3,pkg0 = False
peciagent0,pkg0 = True
pch_chip_cpu_max_temp,pkg0 = False
pch_chip_temp,pkg0 = False
```

amanusk, 2019-04-23:
@GM-Script-Writer-62850, what exactly is broken?

GM-Script-Writer-62850, 2019-04-23:
When i re-open it my saved settings are not applied

amanusk, 2019-04-23:
Thanks, should apply now

amanusk, 2020-01-23:
You can now toggle off sensors in the side menu in version 1.0

closed 2020-01-23 by amanusk