amanusk/s-tui #116: Allow to toggle GPU RPM, and show alongside GPU temperature graph

Valmar33, 2019-04-09:
**Is your feature request related to a problem? Please describe.**

GPU RPM value is shown at the bottom of the sensors list, but isn't selectable in sensors options.

**Describe the solution you'd like**

Add it to be toggleable, and shown besides the GPU temperature graph.

**Additional context**

https://i.imgur.com/obWYsa4.jpg

amanusk, 2019-04-09:
@Valmar33, thanks for opening the feature request.

I have merged a PR with the option to add the Fan sensor as one of the graphs, it should now be controllable as the rest, but will be on a graph on its own.

Out of curiosity, `amdgpu` is an discrete GPU on your system or is it integrated as part of the SOC?

Valmar33, 2019-04-09:
It works! :)

Now all you need is for the rows to be able to shuffled around by the user. Another feature request for another day. :P

The GPU is an RX 580.

Is there a way for `s-tui` to display the GPU name, instead of the driver? Or is this something that has to be done on Mesa's side?

amanusk, 2019-04-09:
One of the goals of the upcoming version is to make everything more modular, so it could be possible to have GPU information for example.
So far we were not focusing on GPUs, mainly because of lack of hardware, but it is somewhere on the roadmap.

closed 2019-04-09 by amanusk