amanusk/s-tui #117: Possible to ignore blatantly wrong fan speeds

GM-Script-Writer-62850, 2019-04-14:
I noticed that from time to time my fan sensor will report a completely wrong impossible value of 19640 RPM, 20 times the fan's rated max speed, this skews the graph's scale making the bars far less useful
[image]
[image]

Also it seems to apply the same max RPM scale to every fan, this is not a issue if all fans are the same model or have the same max RPM, i have a 5000RPM fan (for the VRM) as well a couple 1200RPM RPM fans and couple 1000RPM fan in my system, each fan should have it own separate scaling

amanusk, 2019-04-14:
@GM-Script-Writer-62850 Thanks for testing the new features. Hope they are beneficial to you and all future users.

> I noticed that from time to time my fan sensor will report a completely wrong impossible value of 19640 RPM, 20 times the fan's rated max speed, this skews the graph's scale making the bars far less useful

This might be a problem with how `psutil` reports the RPM, I'll look into it.

> Also it seems to apply the same max RPM scale to every fan, this is not a issue if all fans are the same model or have the same max RPM, i have a 5000RPM fan (for the VRM) as well a couple 1200RPM RPM fans and couple 1000RPM fan in my system, each fan should have it own separate scaling

This will probably not be implemented. The expectation is for all sensors of the same type to be around the same ball park. With all the new graphs coming in the new version, screen real-estate is at a premium as it is.
A scale per-graph is not a priority.

I can also add that the graph axis adjust if current measurements are ways smaller than the maximum recorded. If for instance 65000 RPM was recorded while the usual measurements are lower than 10000, the scale will adjust once the misbehaving bar is far enough in the past.

GM-Script-Writer-62850, 2019-04-14:
I did notice one it passes it returns to normal, usually about the time it get another spike

Also sensors that report a value that are not being displayed on the graph should be ignored for setting the scale, for example i have Auxtin0,Pkg0 that reads 100-127C that i have no clue what it is connected to if anything as well at least the Auxtin1,Pkg0's value is not used for setting the bottom min of the graph as it always reads -128C (last i checked i do not have any LN2 here)
these are set to hidden and my scale for temp is 0 to 127 when the highest graph i have enable is 75C
i could see a static temp ceiling being useful/preferred

that 16,940 value also shows up when using the sensors command

I really only see per graph scaling being useful for fans RPMS

amanusk, 2020-07-11:
I put a filter on negative values for temps and very high RPM for fans in the latest master version

GM-Script-Writer-62850, 2020-07-11:
ot sure where the thermal sensor cold bugs at, but i think most can go a bit below zero, readings of -10 C are probably valid