amanusk/s-tui #120: S-tui fails when switching from 0-workers stress to monitor mode.

nariox, 2019-05-06:
### Step 1: Describe your environment
* Archlinux with 5.0.12 kernel
* Thinkpad T480 
* s-tui version 0.8.3
###  Step 2: Describe the problem:
#### Observed Results:
If I set the stress options to 0 worker for all stress options and switch to stress mode, nothing happens (nothing wrong here), but when I switch back to monitor mode, the program crashes.
#### Expected behavior:
The program either realizes that stress needs at least one worker and prevents stress mode being activated without workers or program realizes no workers were called, so it doesn't try to close workers when none are present
#### Traceback output:
[traceback.txt](https://github.com/amanusk/s-tui/files/3149344/traceback.txt)

amanusk, 2019-05-06:
Thanks @nariox,
I'll push a fix for this in the next version

nariox, 2019-05-09:
Hi @amanusk,
It's really a low priority bug, it only fails when the user does something that doesn't really make sense, but I thought it'd be right to report it anyway. Thank you for the quick and courteous response.

amanusk, 2020-01-23:
Fixed in version 1.0, setting workers to 0 does not activate stress at all.
Please validate if you can.

closed 2020-01-23 by amanusk

nariox, 2020-01-24:
Great, no more crashes :)