amanusk/s-tui #122: Python import module s-tui

mj1595, 2019-05-27:
* System (Intel/AMD/Raspberry Pi etc): Linux(ubuntu)
  * OS version: 18.04.02 LTS
  * s-tui version: 1.0.0-beta1
  * Installation method(pip/PPA/source code): pip3
  
I tried to import s-tui module into python as I need to read power alone and write it into a csv file for a specified period of interval. But unfortunately, I am getting errors while I try to run the module in my code. Is there any way to get functions into my code?

Any help would be highly appreciated. 

Thanks,
Manjunath

amanusk, 2019-05-28:
Hi, Thanks for you interest in the project.
I haven't yet defined a proper API for this, and looking to do so in the future.

To use the call of power in your code you have two options.
1. You can get the values in joules (not Watts)
The call for `rapl_read` returns a list of name tuples
Example:
```python
from s_tui.sources.rapl_read import rapl_read
print(rapl_read())
```
This should print something like this:
`[rapl(label='package-0', current=367508033.0, max=0.0), rapl(label='core', current=171328648.0, max=0.0), rapl(label='uncore', current=30072907.0, max=0.0)]` 

2. You can import the Power source class. Every call for `upate()` will take a new measurement,
then you can call `get_sensors_summary()` to get a dict of  readings in Watts from the last `update`
Here is an example:

```python
from s_tui.sources.rapl_power_source import RaplPowerSource
import time

ps = rapl_power_source = RaplPowerSource()

print(ps.get_sensors_summary())
time.sleep(1)

ps.update()
print(ps.get_sensors_summary())
```
This should print something like this:
`OrderedDict([('package-0', '0'), ('core,Pkg0', '0'), ('uncore,Pkg0', '0')])
OrderedDict([('package-0', '6.1'), ('core,Pkg0', '2.1'), ('uncore,Pkg0', '0.7')])`

Please update if this works for you

mj1595, 2019-05-28:
Thank you very much. It worked. 

I need a small clarification on these as well.

What does pkg0, core and uncore mean? So, if I want to know the total power consumption of my CPU, How can I calculate it with these values?

amanusk, 2019-05-28:
These are values reported by the kernel, labels are also reported by the kernel.

`package-0` usually sums up the power consumption of the entire CPU package.
If you are looking a single number, this is the stat to look at.
 
`core` and `uncore` break it down into.. well.. the power consumptions of the CPU cores and uncore(e.g integrated GPU).

Some CPUs report additional values.

You can probably find more information on this topic by searching for `Intel RAPL`

mj1595, 2019-05-28:
Thank you. This project is really helpful for my project work.  And kudos to you, I am also using your psutil project to get network stats 👍

amanusk, 2019-09-03:
I'll close the issue and open a separate one for tracking API clarification

closed 2019-09-03 by amanusk